intel/media-driver #148: Segmentation fault when playing back h264 video on Ubuntu

xhaihao, 2018-04-18:
I got segmentation fault when using command 'mpv --hwdec --vo vaapi <h264 video file>' to play back h.264 video file. 

> Using hardware decoding (vaapi).
> VO: [vaapi] 3840x2160 vaapi[nv12]
> Segmentation fault (core dumped)ped: 4
>

Backtrace:

> #0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
> #1  0x00007ffff1505f5d in __GI_abort () at abort.c:90
> #2  0x00007ffff154e28d in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff1675528 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:181
> #3  0x00007ffff155564a in malloc_printerr (action=<optimized out>, str=0x7ffff1675860 "double free or corruption (!prev)", ptr=<optimized out>, ar_ptr=<optimized out>) at malloc.c:5426
> #4  0x00007ffff155773e in _int_free (av=0x7fffd0000020, p=<optimized out>, have_lock=0) at malloc.c:4175
> #5  0x00007ffff155c44e in __GI___libc_free (mem=<optimized out>) at malloc.c:3145
> #6  0x00007ffff27be11d in ?? () from /usr/lib/x86_64-linux-gnu/libX11.so.6
> #7  0x00007fffd690f7ac in DdiMedia_PutSurfaceLinuxSW (ctx=0x7fffd0071870, surface=14, draw=0x4e00002, srcx=0, srcy=0, srcw=3840, srch=2160, destx=69, desty=0, destw=1781, desth=1002,
>     cliprects=0x0, number_cliprects=0, flags=32) at /home/xhh/graphics/ex/vaapi/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:936
> #8  0x00007fffd68efddc in DdiMedia_PutSurface (ctx=0x7fffd0071870, surface=14, draw=0x4e00002, srcx=0, srcy=0, srcw=3840, srch=2160, destx=69, desty=0, destw=1781, desth=1002,
>     cliprects=0x0, number_cliprects=0, flags=32) at /home/xhh/graphics/ex/vaapi/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3741
> #9  0x00005555556477df in render_to_screen (mpi=<optimized out>, p=0x5555559a5030) at ../video/out/vo_vaapi.c:530
> #10 flip_page (vo=<optimized out>) at ../video/out/vo_vaapi.c:563
> #11 0x0000555555641cc0 in vo_render_frame_external (vo=vo@entry=0x555555977720) at ../video/out/vo.c:881
> #12 0x00005555556424c8 in vo_thread (ptr=0x555555977720) at ../video/out/vo.c:1016
> #13 0x00007ffff18b47fc in start_thread (arg=0x7fffe525f700) at pthread_create.c:465
> #14 0x00007ffff15e1b5f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
> 
>

XinfengZhang, 2018-05-25:
sorry for late response . will check it

XinfengZhang, 2018-06-06:
will remove DdiMedia_PutSurfaceLinuxSW , and create vp context for DdiMedia_PutSurfaceLinuxHW, is it ok for you @xhaihao

PR #260 (2018-08-20): [VP]Remove sw path and create vp context in DdiMedia_Putsurface
fix #148
commit fe22cf4f (2018-08-20): [VP]Remove sw path and create vp context in DdiMedia_Putsurface to fix #148
closed 2018-08-20 by Sherry-Lin