intel/media-driver #155: [gstreamer-vaapi][vpp] CSC fail.

feiwan1, 2018-04-25:
System Environment
OS Ubuntu 17.04
Kernel 4.10
Hardware Skylake Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
libva tag 2.1.0
intel-media-driver f6283e2
gst-vaapi tag 1.13.90

Reproduce Steps
Use gst cmd to convert NV12 to I420/YUY2/RGBX , and then render picture, but I420 format display black, YUY2/RGBX cmdline fail.
$ gst-launch-1.0 videotestsrc ! vaapipostproc format=2 ! vaapisink
$ gst-launch-1.0 videotestsrc ! vaapipostproc format=4 ! vaapisink
$ gst-launch-1.0 videotestsrc ! vaapipostproc format=7 ! vaapisink

Only YV12 looks good:
$ gst-launch-1.0 videotestsrc ! vaapipostproc format=3 ! vaapisink

FurongZhang, 2018-04-27:
We are working in progress to reproduce it.

FurongZhang, 2018-05-15:
NV12->I420 is related to VP, we will fix it. But for NV12->YUY2 and NV12->RGBX, no VP function is called.

FurongZhang, 2018-05-15:
@feiwan1 Is it possible for you to take a look at NV12->YUY2 and NV12->RGBX?

feiwan1, 2018-05-16:
@FurongZhang , Is all csc should go through VP function in driver? Is there any other way to do csc?

FurongZhang, 2018-06-19:
CSC should go through VP function in driver.

FurongZhang, 2018-09-04:
@feiwan1  This issue is fixed. Please use the latest driver(ww36) to verify that. The fix, please refer to the function in DdiMedia_CreateImage. We will close this if you verify it, thanks a lot in advance!

feiwan1, 2018-09-05:
@wangzj0601, Could you help to verify this patch? Thanks.

FurongZhang, 2018-10-17:
@wangzj0601, is it possible for you to verify it? Thanks a lot in advance!

FurongZhang, 2018-10-17:
Pending on submitter to verify that.

wangzj0601, 2018-12-12:
it can be pass when convert to I420, YV12 or YUY2,  but still fail when convert NV12 to RGBX with driver version ba63dbe

feiwan1, 2020-05-14:
By using glimagesink instead of vaapisink, all case looks good. So close this issue.

Thanks @FurongZhang for fixing this.

closed 2020-05-14 by feiwan1