intel/media-driver #157: Compilation fails with GCC 8.0.1

uartie, 2018-05-02:
libva (master) heads/master-0-gd6fd111e2062 
gmmlib (master) heads/master-0-ga5015343bc93 
intel-media-driver (master) heads/master-0-ge16ab37e9e70 

GCC 8.0.1 is default on recently released Fedora 28.

```
In file included from /opt/media/build/intel-media-driver/media_driver/agnostic/common/renderhal/renderhal.h:36,
                 from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_hal.h:29,
                 from /opt/media/build/intel-media-driver/media_driver/linux/common/cm/cm_surface_manager.h:31,
                 from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_buffer_rt.cpp:29:
/opt/media/build/intel-media-driver/media_driver/agnostic/common/hw/mhw_render.h: In constructor 'MhwRenderInterface::MhwRenderInterface(MhwMiInterface*, PMOS_INTERFACE, MEDIA_SYSTEM_INFO*, uint8_t)':
/opt/media/build/intel-media-driver/media_driver/agnostic/common/hw/mhw_render.h:601:60: error: 'void* memset(void*, int, size_t)' clearing an object of non-trivial type 'MHW_RENDER_ENGINE_L3_CACHE_CONFIG' {aka 'struct _MHW_RENDER_ENGINE_L3_CACHE_CONFIG'}; use assignment or value-initialization instead [-Werror=class-memaccess]
         memset(&m_l3CacheConfig, 0, sizeof(m_l3CacheConfig));
                                                            ^
In file included from /opt/media/build/intel-media-driver/media_driver/agnostic/common/renderhal/renderhal.h:36,
                 from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_hal.h:29,
                 from /opt/media/build/intel-media-driver/media_driver/linux/common/cm/cm_surface_manager.h:31,
                 from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_buffer_rt.cpp:29:
/opt/media/build/intel-media-driver/media_driver/agnostic/common/hw/mhw_render.h:63:16: note: 'MHW_RENDER_ENGINE_L3_CACHE_CONFIG' {aka 'struct _MHW_RENDER_ENGINE_L3_CACHE_CONFIG'} declared here
 typedef struct _MHW_RENDER_ENGINE_L3_CACHE_CONFIG
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video.dir/build.make:76: media_driver/CMakeFiles/iHD_drv_video.dir/agnostic/common/cm/cm_buffer_rt.cpp.o] Error 1
```

oliver-sang, 2018-05-03:
thanks for reporting the compiling issue under new compiler. We will look at this.

chivakker, 2018-05-24:
Similar issues are breaking compilation with gcc-8.1.1

i.e.

/media-driver/media_driver/agnostic/common/codec/hal/codechal_kernel_hme.cpp:44:50: error: ‘void* memset(void*, int, size_t)’ clearing an object of non-trivial type ‘struct CodechalKernelHme::CurbeParam’; use assignment or value-initialization instead [-Werror=class-memaccess]
     memset(&m_curbeParam, 0, sizeof(m_curbeParam));

oliver-sang, 2018-05-25:
the situation is worse than we first estimated, especially Non-POD memset problem, we've got hundreds warning about it, which each one should be check and fix. team is keeping on the fix.

jwang11, 2018-06-11:
In latest clear linux, which upgraded to gcc 8.x, I also met a lot of compilation issues. I'll provide a PR to fix them.

chivakker, 2018-06-11:
@jwang11, there's already people working on it.  see @oliver-sang comment above.

jwang11, 2018-06-12:
@chivakker Ok, sounds good. is there a timeline to fix it because I'd like integrate it into Clear Linux ASAP? Just notice it has pending for month. From error logs, there are many types of error. Is it possible to split the big issue to a bunch of small ones, such as Non-Pod memset,  delete-non-virtual-dtor. I can extend help on some of them to expedite.

oliver-sang, 2018-06-12:
we plan to push first version this week. but first it will be for fedora28/gcc8.0.1.

oliver-sang, 2018-06-15:
we have a series fix and 97916802ee0964b4a2b03237ce0606dd2b35351f is the last one.
Could you please help check on your platform to see whether it addresses your problem? Thanks

uartie, 2018-06-15:
Latest fixes work for me on F28 w/ GCC 8.0.1

uartie, 2018-06-15:
Also, F28 w/ GCC 8.1.1 worked for me.

jwang11, 2018-06-16:
Using latest media-driver, in ClearLinux with gcc 8.1.1, I can still observe some other compile issues, such as
/.../media_vaapi/media-driver/cmrtlib/agnostic/share/cm_printf_host.cpp:708:43: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
             CM_PRINTF(m_streamOut, tkn, *((double*)&(header->scalar64)));

/.../media_vaapi/media-driver/cmrtlib/agnostic/share/cm_printf_host.cpp:786:64: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
             CM_PRINTF(m_streamOut,tkn, m_args[0], m_args[1], *((float*          )&(header->scalar64)));
                        
In addition, looks libva release 2.1 that ClearLinux already integrated can't work well with latest media-driver, 
/../media_vaapi/media-driver/media_driver/linux/common/ddi/media_libva.h:213:5: error: ‘VASliceParameterBufferHEVCExtension’ does not name a type; did you mean ‘VASliceParameterBufferHEVC’?
     VASliceParameterBufferHEVCExtension          *pVASliceParaBufHEVCRext;

chivakker, 2018-06-19:
@jwang11 those errors are not related to gcc-8.x.x.  Please see issues #197 and #198

jwang11, 2018-06-20:
@chivakker you are right, the errors should be from CMAKE_CXX_FLAGS of clear linux. I just extract it out from CMakeCache.txt as below
```
CMAKE_CXX_FLAGS:STRING=-g -O3 -feliminate-unused-debug-types  -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=32 -Wformat -Wformat-security -Wl,--copy-dt-needed-entries -m64  -fasynchronous-unwind-tables -Wp,-D_REENTRANT -ftree-loop-distribute-patterns -Wl,-z -Wl,now -Wl,-z -Wl,relro -fno-semantic-interposition -ffat-lto-objects  -fno-signed-zeros -fno-trapping-math  -fassociative-math -Wl,-sort-common -Wl,--enable-new-dtags -fvisibility-inlines-hidden -Wl,--enable-new-dtags
```

oliver-sang, 2018-06-27:
Hello guys, are we clean now?