intel/media-driver #158: Failed to do trancode of h265 main profile  with FFmpeg on CFL

xhaihao, 2018-05-09:
CFL is supported since commit 9b0caac9692f7e47e2cac01853390f147f9ebfa1, however I failed to do transcode on CFL:

software stacks:
libva: d6fd111e2062bb4732db8a05ed55fc01771087b4
media_driver: 5c334b3e1fb7f5c0a7c2218fa912278c081f65e5
gmmlib: a5015343bc932e39747c57ea5dec0cbf28685465
ffmpeg: 0736f32a4facddbd953977ca614a3ee6d8a6e1d7

command line: ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i h265-main.h265 -c:v hevc_vaapi -profile:v main output.h265

Error message (use debug version of media_driver)
[MHW]:  CRITICAL - AddData:53: Data will not fit within the this memory block or state heap
[CODEC]:CRITICAL - AddCurbeToStateHeap:5337: MOS returned error, eStatus = 0x2
[CODEC]:CRITICAL - Encode8x8PBMbEncKernel:7213: MOS returned error, eStatus = 0x2
[CODEC]:CRITICAL - EncodeKernelFunctions:9573: MOS returned error, eStatus = 0x2
[CODEC]:CRITICAL - ExecuteKernelFunctions:3015: MOS returned error, eStatus = 0x2
[CODEC]:CRITICAL - ExecuteEnc:4273: ENC failed.
[CODEC]:CRITICAL - Execute:567: MOS returned error, eStatus = 0x2
[LIBVA]:CRITICAL - EncodeInCodecHal:361: DDI:Failed in Codechal!
[LIBVA]:CRITICAL - EndPicture:80: DDI:DdiEncode_EncodeInCodecHal return failure.
[hevc_vaapi @ 0x561371b80e50] Failed to end picture encode issue: 24 (internal encoding error).
[hevc_vaapi @ 0x561371b80e50] Encode failed: -5.

XinfengZhang, 2018-05-10:
thanks for the reporting, will fix it ASAP

XinfengZhang, 2018-05-17:
P frame issue, CFL HEVC only support I B B1 B2.

closed 2018-06-06 by XinfengZhang