intel/media-driver #160: [SKL][AVC DEC] AVC decoder sync surface fail

mypopydev, 2018-05-16:
Get the test sample from: https://trac.ffmpeg.org/ticket/7200

Use the command: 

./ffmpeg_g -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_flags allow_profile_mismatch -i test.ts -f null /dev/null
ffmpeg version N-91095-g64f59a21b3 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.2.0-8ubuntu3.2)
  configuration: --disable-optimizations --disable-stripping --enable-libx264 --enable-libx265 --enable-gpl --samples=../fate-suite --enable-libpulse --enable-libwebp
  libavutil      56. 18.102 / 56. 18.102
  libavcodec     58. 19.101 / 58. 19.101
  libavformat    58. 13.102 / 58. 13.102
  libavdevice    58.  4.100 / 58.  4.100
  libavfilter     7. 22.100 /  7. 22.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
Input #0, mpegts, from '../../../Videos/test.ts':
  Duration: 00:00:08.84, start: 1.400000, bitrate: 2136 kb/s
  Program 1 
    Metadata:
      service_name    : Service01
      service_provider: FFmpeg
    Stream #0:0[0x100]: Video: h264 (Baseline) ([27][0][0][0] / 0x001B), yuv420p(tv, bt470bg/bt470bg/smpte170m, progressive), 1280x720, 23.98 tbr, 90k tbn, 180k tbc
    Stream #0:1[0x101]: Audio: aac (LC) ([15][0][0][0] / 0x000F), 44100 Hz, mono, fltp, 98 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> wrapped_avframe (native))
  Stream #0:1 -> #0:1 (aac (native) -> pcm_s16le (native))
Press [q] to stop, [?] for help
[h264 @ 0x55615971ed00] Using possibly-incompatible profile 100 instead.
    Last message repeated 1 times
Output #0, null, to '/dev/null':
  Metadata:
    encoder         : Lavf58.13.102
    Stream #0:0: Video: wrapped_avframe, nv12, 1280x720, q=2-31, 200 kb/s, 23.98 fps, 23.98 tbn, 23.98 tbc
    Metadata:
      encoder         : Lavc58.19.101 wrapped_avframe
    Stream #0:1: Audio: pcm_s16le, 44100 Hz, mono, s16, 705 kb/s
    Metadata:
      encoder         : Lavc58.19.101 pcm_s16le
**[AVHWFramesContext @ 0x7fef040561c0] Failed to sync surface 0x16: 23 (internal decoding error).
[h264 @ 0x5561598baf40] Failed to transfer data to output frame: -5.
Error while processing the decoded data for stream #0:0
Conversion failed!**

XinfengZhang, 2018-05-16:
bit stream of frame 176 is not completely. so it should be expectation to return error, you can call vaQuerySurfaceError to check error details.

mypopydev, 2018-05-23:
@XinfengZhang thanks the suggestion, will add vaQuerySurfaceError to get the decoding details error.