intel/media-driver #164: [SKL] vaAcquireSurfaceHandle() failed (the requested function is not implemented)

enihcam, 2018-05-20:
Intel(R) Core(TM) i5-7200U CPU @ 2.50GHz
Archlinux (with all packages up to latest)
intel-media-driver ver: 2018.Q2.1

```
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[ffmpeg] AVHWFramesContext: Failed to create surface: 14 (the requested RT Format is not supported).
[ffmpeg] AVHWFramesContext: Unable to allocate a surface from internal buffer pool.
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
AO: [alsa] 48000Hz stereo 2ch double
Using hardware decoding (vaapi).
VO: [gpu] 1920x804 vaapi[nv12]
[vo/gpu/vaapi-egl] vaAcquireSurfaceHandle() failed (the requested function is not implemented)
AV: 00:09:52 / 02:16:01 (7%) A-V:  0.000 Cache: 576s+409MB
```

```
libva info: VA-API version 1.1.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_1
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.1 (libva 2.1.0)
vainfo: Driver version: Intel iHD driver - 2.0.0
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD

```

XinfengZhang, 2018-05-21:
hi , no vaAcquireSurfaceHandle support, only vaAcquireBufferHandle, could you add more details, especially libva version (commit id)

enihcam, 2018-05-21:
@XinfengZhang what details do you need? as you can see from my previous message, the version is
`vainfo: VA-API version: 1.1 (libva 2.1.0)`

XinfengZhang, 2018-05-23:
want to know the commit id of libva, I can check whether the vaAcquireSurfaceHandle is included in the va_backend.h. current libva tips has no such definition.

enihcam, 2018-05-23:
@XinfengZhang https://github.com/intel/libva/commit/417b0184d4a767721e9b0a5f1597907707f545cf

XinfengZhang, 2018-05-24:
@enihcam  only vaAcquireBufferHandle https://github.com/intel/libva/blob/master/va/va_backend.h#L427 . no vaAcquieSurfaceHandle

XinfengZhang, 2018-05-24:
https://github.com/intel/libva/blob/master/va/va.h#L3390

XinfengZhang, 2018-06-06:
#177

XinfengZhang, 2019-08-27:
vaExportSurfaceHandle already implemented with DdiMedia_ExportSurfaceHandle

closed 2019-08-27 by XinfengZhang