intel/media-driver #165: [gstreamer-vaapi][h264enc] Fail to encode with qtmux for H264 encoding

TheanSiew, 2018-05-21:
[System Environment]
Platform : APL
Kernel     : 4.17-rc3 
libva        : master branch d6fd111e2062bb4732db8a05ed55fc01771087b4
gst-vaapi : 1.14.0

[Defect description]
Encoding H264 from raw video will fail if qtmux is called. Removing qtmux from the command line then H264 encoding works fine. 

[Reproduce Steps]
1. Command used:
	$ gst-launch-1.0 -v filesrc location= /media/RAWS/NV12_Prog_Planar_720x480.yuv ! videoparse width=720 height=480 format=nv12 ! vaapih264enc ! qtmux ! filesink location=test.264
2. Error information as below:
	Setting pipeline to PAUSED ...

	Pipeline is PREROLLING ...

	Got context from element 'vaapiencodeh264-0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayGLX\)\ vaapidisplayglx0";

	/GstPipeline:pipeline0/GstVideoParse:videoparse0/GstRawVideoParse:inner_rawvideoparse.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)720, height=(int)480, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt601, framerate=(fraction)25/1

	/GstPipeline:pipeline0/GstVideoParse:videoparse0.GstGhostPad:src: caps = video/x-raw, format=(string)NV12, width=(int)720, height=(int)480, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt601, framerate=(fraction)25/1

	/GstPipeline:pipeline0/GstVideoParse:videoparse0.GstGhostPad:src.GstProxyPad:proxypad1: caps = video/x-raw, format=(string)NV12, width=(int)720, height=(int)480, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt601, framerate=(fraction)25/1

	ERROR: from element /GstPipeline:pipeline0/GstVideoParse:videoparse0/GstRawVideoParse:inner_rawvideoparse: Internal data stream error.

	Additional debug info:

	../../../../gstreamer-1.14.0/libs/gst/base/gstbaseparse.c(3611): gst_base_parse_loop (): /GstPipeline:pipeline0/GstVideoParse:videoparse0/GstRawVideoParse:inner_rawvideoparse:

	streaming stopped, reason not-negotiated (-4)

	ERROR: pipeline doesn't want to preroll.

	Setting pipeline to NULL ...

	Freeing pipeline ...

XinfengZhang, 2018-05-23:
will check it.

sreerenjb, 2018-05-23:
My test cases working fine. Is this happening only with a specific sample?
Please enable GST_DEBUG=3 and check what is the exact error.
Also, check whether providing "framerate=25/1" in videoparse fixing the issue. BTW, provide h264parse after the encoder.

TheanSiew, 2018-05-24:
I tried another raw NV12 file also having the same issue. Previously @XinfengZhang mentioned that he saw issue when qtmux is being used too. Does @XinfengZhang still able to reproduce this issue?

I had tried both commands as you suggested but still having the same error:

> $ gst-launch-1.0 filesrc location=/home/root/videos/raw/NV12_Prog_Planar_720x480.yuv ! videoparse width=720 height=480 format=nv12 framerate=25/1 ! vaapih264enc ! qtmux ! filesink location=test.264
> $ gst-launch-1.0 filesrc location=/home/root/videos/raw/NV12_Prog_Planar_720x480.yuv ! videoparse width=720 height=480 format=nv12 framerate=25/1 ! vaapih264enc ! h264parse ! qtmux ! filesink location=test.264

Below are the output when GST_DEBUG=3 is turned on:

> Setting pipeline to PAUSED ...
> 0:00:00.069815914  1019       0x8ad010 WARN                 basesrc gstbasesrc.c:3583:gst_base_src_start_complete:<filesrc0> pad not activated yet
> Pipeline is PREROLLING ...
> Got context from element 'vaapiencodeh264-0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayDRM\)\ vaapidisplaydrm1";
> 0:00:00.072828607  1019       0x882d40 WARN            videoencoder gstvideoencoder.c:678:gst_video_encoder_setcaps:<vaapiencodeh264-0> rejected caps video/x-raw, format=(string)NV12, width=(int)720, height=(int)480, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt601, framerate=(fraction)25/1
> 0:00:00.073172508  1019       0x882d40 WARN            videoencoder gstvideoencoder.c:678:gst_video_encoder_setcaps:<vaapiencodeh264-0> rejected caps video/x-raw, format=(string)NV12, width=(int)720, height=(int)480, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt601, framerate=(fraction)25/1
> 0:00:00.073416932  1019       0x882d40 WARN            videoencoder gstvideoencoder.c:678:gst_video_encoder_setcaps:<vaapiencodeh264-0> rejected caps video/x-raw, format=(string)NV12, width=(int)720, height=(int)480, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt601, framerate=(fraction)25/1
> 0:00:00.073572859  1019       0x882d40 WARN               baseparse gstbaseparse.c:3611:gst_base_parse_loop:<inner_rawvideoparse> error: Internal data stream error.
> 0:00:00.073594101  1019       0x882d40 WARN               baseparse gstbaseparse.c:3611:gst_base_parse_loop:<inner_rawvideoparse> error: streaming stopped, reason not-negotiated (-4)
> ERROR: from element /GstPipeline:pipeline0/GstVideoParse:videoparse0/GstRawVideoParse:inner_rawvideoparse: Internal data stream error.
> Additional debug info:
> ../../../../gstreamer-1.14.0/libs/gst/base/gstbaseparse.c(3611): gst_base_parse_loop (): /GstPipeline:pipeline0/GstVideoParse:videoparse0/GstRawVideoParse:inner_rawvideoparse:
> streaming stopped, reason not-negotiated (-4)
> ERROR: pipeline doesn't want to preroll.
> Setting pipeline to NULL ...
> 0:00:00.074005408  1019       0x882d40 WARN            videoencoder gstvideoencoder.c:678:gst_video_encoder_setcaps:<vaapiencodeh264-0> rejected caps video/x-raw, format=(string)NV12, width=(int)720, height=(int)480, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt601, framerate=(fraction)25/1
> 0:00:00.074276275  1019       0x882d40 WARN            videoencoder gstvideoencoder.c:678:gst_video_encoder_setcaps:<vaapiencodeh264-0> rejected caps video/x-raw, format=(string)NV12, width=(int)720, height=(int)480, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt601, framerate=(fraction)25/1
> 0:00:00.074515068  1019       0x882d40 WARN            videoencoder gstvideoencoder.c:678:gst_video_encoder_setcaps:<vaapiencodeh264-0> rejected caps video/x-raw, format=(string)NV12, width=(int)720, height=(int)480, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt601, framerate=(fraction)25/1
>

jcescamilla, 2018-05-26:
I'm not able to reproduce this issue with the latest code. Using master branch on all components bellow.
libva
    commit: b3be72a5a110880f70626d7c3bed953cdde124b2
media-driver
    commit: 5ae54891f368a37bd25d1ba83cac6b820b61aad4
gmmlib
    commit: 40b45bd6109a6a485b4a721686fc32ba7307a548
gstreamer
    commit: cc869065a59dfef7b6bc6cf6f6df270f0d371282
gst-plugins-base
    commit: 9158f9f7eb5039ae67b3463f5dbac8d3ebb540ac
gst-plugins-bad
    commit: 2d98a5c1d7854d5d15678945f84b2960a52f75f8
gst-plugins-good
    commit: 364a921d3f10c54d403b5f35afbbdb405dbca819
gstreamer-vaapi
    commit: 705b37844e45f3cf37581ae39128b7a488fd4faa

closed 2018-06-21 by XinfengZhang

talksik, 2023-11-02:
I had to use h264parse right after vaapih264enc. Ahh isn't the gstreamer world just great :(