intel/media-driver #169: Strict memory alignment and padding requirement makes it hard to negotiate dmabuf memory between drivers

sreerenjb, 2018-05-24:
Importing dmabuf backed memory to vaapi surface is a very important and already supported feature in intel-vaapi-driver. But this is not working with media-driver.

For eg:
The application needs a YUY2 surface of size 640x480 (width and height are 32 bit aligned) and video4linux driver allocates buffers of size 614400 (pitch * height = 1280 * 480).  But when we try to import this buffer to vaapi surface,media-driver will fail because GMMLib requires a memory of size 618496 bytes. See the code here:https://github.com/intel/media-driver/blob/master/media_driver/linux/common/ddi/media_libva_util.cpp#L624

The reason is that YUY2 surfaces have more padding restrictions and vasurface requires bigger size to operate [According to spec].
But unfortunately, the video4linux APIs won't allow creating a memory of specific padding or size.
If we want to import the dmabuf-backed memory created by v4l2, we should be able to accept whatever size it comes with.

What I can see is that intel-vaapi-driver doesn't have any memory restrictions and it accept whatever size the underline dmabuf memory possess. It may be a wrong approach if we follow the spec, but works well for many customers :)

Is there any workaround possible for this scenario?

XinfengZhang, 2018-11-19:
the restriction is because we need to add 1 line , then align with 4096. 
now, we removed restriction of UserPTR by commit e38b7e4b.

XinfengZhang, 2019-08-27:
keep it , just because the YUY2 linear surface size is special

XinfengZhang, 2020-05-25:
the alignment request is handled by https://github.com/intel/gmmlib, it should move to gmmlib

closed 2022-06-27 by XinfengZhang