intel/media-driver #170: BUILD_ALONG_WITH_CMRT=0 still builds cmrtlib subdirectory

chivakker, 2018-05-24:
It is my understanding that driver should work without CMRT compiled in.  I am seeing a bunch of compilation errors even if I pass 

cmake ../media-driver -DCMAKE_INSTALL_PREFIX=/usr -DMEDIA_VERSION="2.0.0" -DBUILD_ALONG_WITH_CMRTLIB=0 -DBS_DIR_GMMLIB=`pwd`/../gmmlib/Source/GmmLib/ -DBS_DIR_COMMON=`pwd`/../gmmlib/Source/Common/ -DBS_DIR_INC=`pwd`/../gmmlib/Source/inc/ -DBS_DIR_MEDIA=`pwd`/../media-driver -Wno-class-memaccess

As #157 happens, I cannot move forward with compiling iHD.

oliver-sang, 2018-05-25:
the BUILD_ALONG_WITH_CMRTLIB is actually obsoleted. we will remove this option soon.

#157 is because our driver is not ready for gcc8. team is working on the fix now.

oliver-sang, 2018-06-15:
BUILD_ALONG_WITH_CMRT has been removed by c3e13c175d12790ab70c4de3521e47f62be780de
and gcc8x issue please track in #157

closed 2018-06-15 by oliver-sang