intel/media-driver #176: Can we use the media-driver to decode hevc video up to 8192x8192 ?

jinlongzhou, 2018-06-01:
I used to  take vaapi-intel-driver[https://github.com/intel/intel-vaapi-driver](url) as the backend of the libva , but i found that it can not decode hevc videos whoes size over 4k.  i dont think this is the problem of the libva, but it is the problem of vaapi-intel-driver.  So here is the question, if i replace the "vaapi-intel-driver" with "media-driver" as the backend of libva, can we decode hevc video up to 8k?

XinfengZhang, 2018-06-06:
which platform? HW support 8K decode from BDW+, but we just export 4K resolution caps.

XinfengZhang, 2018-06-11:
it's a small change, you can ignore the resolution report from vaQueryConfigAttributes and vaQuerySurfaceAttributes , call decode directly. if you need it, I can file a PR for you to export 8K capabilities.

jinlongzhou, 2018-06-11:
thank you , i really need it

XinfengZhang, 2018-06-20:
https://github.com/XinfengZhang/media-driver/tree/hevc_8k
you can refer the head  commit of this branch.  thanks

XinfengZhang, 2018-07-31:
@jinlongzhou  could you close this issue?

closed 2018-11-19 by XinfengZhang