intel/media-driver #178: [gstreamer-msdk][mjpeg decode] mjpeg decoding fail with error message "gst_msdkdec_init_decoder:<msdkmjpegdec0> Init failed"

wangzj0601, 2018-06-05:
Test Env:
============================================
Platform:                  APL/KBl/SKL
Arch:          		x86_64
Linux release           16.04
Kernel:			4.12.0-rc2 
libva                         https://github.com/01org/libva.gitb
                                commit 3be72a5a110880f70626d7c3bed953cdde124b2
media_driver		https://github.com/intel/media-driver
                                commit 1c2b0615d749c45c07f9aee6586774816989c5b3
MediaSDK:               https://github.com/Intel-Media-SDK/MediaSDK
                                commit 7c2b069dce7bed268806f680412a2f3b09a52ce9
gst-bad 		        master branch 0bdcf51baf77926b4f29c01a2fdf133c13aad62e

Reproduce Steps:
============================================
take one case for example
1. build enc as above lists
2. gst-launch-1.0 filesrc location=/media/JPEG_RefCompare/0795image04_1152x864.jpg '!' jpegparse '!' msdkmjpegdec '!' videoconvert '!' video/x-raw,format=I420 '!' checksumsink2 frame-checksum=FALSE file-checksum=FALSE plane-checksum=FALSE dump-output=TRUE dump-location=./0795image04_1152x864.jpg.I420
3. error message as below
Got context from element 'msdkmjpegdec0': gst.msdk.Context=context, gst.msdk.Context=(GstMsdkContext)"\(GstMsdkContext\)\ msdkcontext0";
Redistribute latency...
0:00:00.055560235  5939       0xd7bcf0 ERROR                msdkdec gstmsdkdec.c:337:gst_msdkdec_init_decoder:<msdkmjpegdec0> Init failed (undeveloped feature)
Caught SIGSEGV

Failed cases
decode_jpeg/0795image04_1152x864.jpg 10015.68 FAIL
decode_jpeg/0795image04_1280x1024.jpg 10017.11 FAIL
decode_jpeg/27022005_198.sized.jpg 10016.63 FAIL
decode_jpeg/capture20.jpg 10016.81 FAIL
decode_jpeg/capture21.jpg 10016.65 FAIL
decode_jpeg/capture22.jpg 10016.60 FAIL

wangzj0601, 2018-06-05:
This issue can not be duplicated with close source MediaServerStudio2017.

XinfengZhang, 2018-06-13:
suppose it should be a open media sdk issue @onabiull

dmitryermilov, 2018-06-19:
Please attach 0795image04_1152x864.jpg

uartie, 2018-06-19:
@dmitryermilov AFAICT, this happens with any jpeg input file.

uartie, 2018-06-19:
...for example, I generated jpeg via:  `GST_VAAPI_ALL_DRIVERS=1 LIBVA_DRIVER_NAME=iHD gst-launch-1.0 videotestsrc num-buffers=10 ! video/x-raw,format=NV12 ! msdkjpegenc ! jpegparse ! filesink location=test.jpg` and test.jpg fails to decode as in initial description.

dmitryermilov, 2018-06-21:
**It's a gst plug-ins issue**, see https://github.com/GStreamer/gst-plugins-bad/blob/0bdcf51baf77926b4f29c01a2fdf133c13aad62e/sys/msdk/gstmsdkmjpegdec.c#L72.

Instead of calling MSDK DecodeHeader() function and getting actual parameters from jpeg file, the code hardcodes  scan type to non-interleaved **while actually test.jpg has interleaved scan type**. MFX_SCANTYPE_NONINTERLEAVED is not supported in HW MSDK library:
https://github.com/Intel-Media-SDK/MediaSDK/blob/9033e416ab88b08fdd15ca1c9cf6c13d2b4894a6/_studio/mfx_lib/decode/mjpeg/src/mfx_mjpeg_dec_decode.cpp#L1039. The case 'worked' with close source MSDK is because close source MSDK had software fallback for missing HW functionality.

The hack in msdkmjpegenc should be removed.

rlim77, 2018-07-12:
I don't think this is a MSDK nor a media-driver issue. This is most probably an issue with the plugins. gst-mfx topic branch seems to have the same issue. See here:
https://github.com/intel/gstreamer-media-SDK/issues/110

Issue seems to be related to JPEG 4:4:4 decode. And sample_decode using the latest upstream ingredients can decode this no problem.

gst-mfx MASTER can also decode this albeit with explicit capsfiltering specified: 
gst-launch-1.0 -v filesrc location=/home/root/JPEG_RefCompare/A_444_JPEG_File.jpg ! jpegparse ! image/jpeg, width=1152, height=864 ! mfxjpegdec ! video/x-raw, format=BGRA ! mfxsink

wangzj0601, 2018-07-12:
yeah, this is a gst-bad plug-in issue, and it can not duplicated with iHD+gst-vaapi, so close it

closed 2018-07-12 by wangzj0601

sreerenjb, 2018-07-12:
That hack was added to support the closed source msdk, also there were no DecodeHeader implementation at that time. Current git master always uses DecodeHeader API to fill the videoparam and it also respects the upstream parameters (eg: from demuxers).