intel/media-driver #181: [gstreamer-vaapi][vaapipostproc] BOB deinterlace does not take effect

TheanSiew, 2018-06-06:
[System Environment]
Platform : APL
Kernel : 4.17-rc3
libva : master branch d6fd111e2062bb4732db8a05ed55fc01771087b4
gst-vaapi : 1.14.0

[Defect description]
Deinterlace does not take effect when deinterlace mode is turned on and BOB deinterlace method is specified in vaapipostproc. 

[Reproduce Steps]
$ gst-launch-1.0 filesrc location=/home/root/videos/H264_Main@L2.2_0720x0576_03.9Mbps_10.0fps_Interlaced_Sound_ToTheLimit@.mp4 ! qtdemux ! h264parse ! vaapih264dec ! vaapipostproc deinterlace-mode=1 deinterlace-method=1 ! vaapisink

The output is the same when explicitly turned off deinterlace mode:
$ gst-launch-1.0 filesrc location=/home/root/videos/H264_Main@L2.2_0720x0576_03.9Mbps_10.0fps_Interlaced_Sound_ToTheLimit@.mp4 ! qtdemux ! h264parse ! vaapih264dec ! vaapipostproc deinterlace-mode=2 ! vaapisink

XinfengZhang, 2018-06-19:
reproduced. will fix it ASAP

Jister, 2018-10-10:
gstreamer-vaapi has fix the issue by commit ee27377cb38293f8e4f807ad4d0eaaf6d9740174

FurongZhang, 2018-10-17:
Since gstreamer-vaapi has fixed the issue by commit ee27377cb38293f8e4f807ad4d0eaaf6d9740174, close it. @TheanSiew Please verify it. Thanks a lot in advance!

FurongZhang, 2018-10-18:
The submitter TheanSiew has already verified that this issue is fixed.