intel/media-driver #186: HEVC CBR encode bitrate out of control when using ffmpeg-vaapi to do encode

FocusLuo, 2018-06-12:
Test cmdline: 
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -s:v 720x480 -i ./720x480p.yuv -vf format=nv12,hwupload -c:v hevc_vaapi -g 30 -bf 3 -b:v 2M -maxrate:v 2M -quality 4 -vframes 300 -y ./hevce_cbr_720x480_gop30_refdist3_bitrate2M.h265

The CBR bitrate can work well on intel-vaapi-driver(i965), but failed on intel-media-driver(iHD) by using cmdline as up
1) on iHD driver log output: 
frame=  300 fps=0.0 q=-0.0 Lsize=    2542kB time=00:00:11.96 bitrate=1741.2kbits/s speed=13.4x
2) on i965 driver log output:
frame=  300 fps=0.0 q=-0.0 Lsize=    3017kB time=00:00:11.96 bitrate=2066.3kbits/s speed=  12x

uartie, 2018-06-12:
#187 fixes this for me

FocusLuo, 2018-08-08:
the issue fixed and verified

closed 2018-08-08 by FocusLuo