intel/media-driver #189: [ffmpeg-vaapi][mpeg2 encode] mpeg2 encode failed when using ffmpeg-vaapi to do encode

FocusLuo, 2018-06-14:
On SKL, the cmd line can work on intel-vaapi-driver(i965), but failed on intel-media-driver(iHD)
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -y -s:v 1920x1080 -i ./encoderbitstreams//1080p_blue_sky.yuv -vf format=nv12,hwupload -c:v mpeg2_vaapi -g 30 -bf 1 -global_quality 20 test.m2v
login shown as below:
[rawvideo @ 0x3603480] Estimating duration from bitrate, this may be inaccurate
Input #0, rawvideo, from '/root/media_streams//encoderbitstreams//1080p_blue_sky.yuv':
  Duration: 00:00:08.68, start: 0.000000, bitrate: 622080 kb/s
    Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p, 1920x1080, 622080 kb/s, 25 tbr, 25 tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> mpeg2video (mpeg2_vaapi))
Press [q] to stop, [?] for help
[mpeg2_vaapi @ 0x3629c00] Warning: some packed headers are not supported (want 0x3, got 0).
Output #0, mpeg2video, to 'test.m2v':
  Metadata:
    encoder         : Lavf58.17.100
    Stream #0:0: Video: mpeg2video (mpeg2_vaapi) (Main), vaapi_vld, 1920x1080, q=2-31, 200 kb/s, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.19.105 mpeg2_vaapi
[mpeg2_vaapi @ 0x3629c00] Failed to end picture encode issue: 24 (internal encoding error).
[mpeg2_vaapi @ 0x3629c00] Encode failed: -5.
Video encoding failed
Conversion failed!

mypopydev, 2018-06-14:
I think the issue same as #82

XinfengZhang, 2018-07-10:
still have issues with f_code setting, will file a PR to change the check of f_code value

jsunintel, 2018-07-12:
https://github.com/intel/media-driver/pull/227

jsunintel, 2018-07-12:
Cherry picked a patch to this repository.

FocusLuo, 2018-08-08:
This issue had been fixed

closed 2018-08-08 by FocusLuo