intel/media-driver #190: [ffmpeg-vaapi][vp8 decode] almost all vp8 decode test cases failed when using ffmpeg-vaapi

FocusLuo, 2018-06-14:
These test cases can work on intel-vaapi-driver
One CMD line shown as below:
ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -v verbose -y -i ./vp8_vpg/vp8_rev0_I_frame_H_Anim.webm -pix_fmt yuv420p -f md5 ffmpeg_vaapi.md5

The log info shown as below:
ffmpeg version N-91230-gb748772 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.9) 20160609
  configuration: --enable-vaapi --prefix=/opt/X11R7/ffmpeg_mainline
  libavutil      56. 18.102 / 56. 18.102
  libavcodec     58. 19.105 / 58. 19.105
  libavformat    58. 17.100 / 58. 17.100
  libavdevice    58.  4.100 / 58.  4.100
  libavfilter     7. 25.100 /  7. 25.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
Input #0, matroska,webm, from '/media//vp8_vpg/vp8_rev0_I_frame_H_Anim.webm':
  Metadata:
    encoder         : vpxenc v0.9.6
  Duration: 00:00:00.03, start: 0.000000, bitrate: 146 kb/s
    Stream #0:0(eng): Video: vp8, 1 reference frame, yuv420p(progressive), 32x32, SAR 1:1 DAR 1:1, 30 fps, 30 tbr, 1k tbn, 1k tbc (default)
[AVHWDeviceContext @ 0x25fba80] Opened VA display via DRM device /dev/dri/renderD128.
[AVHWDeviceContext @ 0x25fba80] libva: VA-API version 1.1.0
[AVHWDeviceContext @ 0x25fba80] libva: va_getDriverName() returns 0
[AVHWDeviceContext @ 0x25fba80] libva: User requested driver 'iHD'
[AVHWDeviceContext @ 0x25fba80] libva: Trying to open /opt/X11R7/vpg_driver/lib/dri//iHD_drv_video.so
[AVHWDeviceContext @ 0x25fba80] libva: Found init function __vaDriverInit_1_1
[AVHWDeviceContext @ 0x25fba80] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x25fba80] Initialised VAAPI connection: version 1.1
[AVHWDeviceContext @ 0x25fba80] Unknown driver "Intel iHD driver - 2.0.0", assuming standard behaviour.
Stream mapping:
  Stream #0:0 -> #0:0 (vp8 (native) -> rawvideo (native))
Press [q] to stop, [?] for help
[AVHWFramesContext @ 0x7f41e0000a80] Failed to sync surface 0xb: 34 (HW busy now).
[vp8 @ 0x25fd0c0] Failed to transfer data to output frame: -5.
Error while processing the decoded data for stream #0:0
Conversion failed!

mypopydev, 2018-06-14:
I think same as the issue #57

XinfengZhang, 2018-07-10:
even after merging #86  still have some issue,  ffmpeg create 2 context,  register reference surface 0-15 to context 0, register reference surface 16-31 to context 1,  after decode first frame to surface 15, it call vaDestoryContext to destroy context 0 and continue to decode another frame with context1. still use 15 as the reference. it is an invalid reference frame.

uartie, 2018-10-17:
Is this still an issue?  I can't reproduce on today's mainline stack.

uartie, 2018-11-14:
Yes, I'm seeing a lot of failures on my end now.  Not sure what changed.

uartie, 2018-11-14:
Setting `-threads 1` before `-i` option works

closed 2020-04-27 by FocusLuo