intel/media-driver #191: [ffmpeg-vaapi][vp8 encode] vp8 encode failed when using ffmpeg-vaapi

FocusLuo, 2018-06-14:
These test cases can work on intel-vaapi-driver
One CMD line shown as below:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -y -s:v 1920x1080 -i ./1080p_blue_sky.yuv -vf format=nv12,hwupload -c:v vp8_vaapi -loop_filter_level 16 -loop_filter_sharpness 12 -g 30 -global_quality 40 1080p_1920x1080_i420_looplevel16_loopsharpness12_bf1_gop30_gquality40_fps30.ivf

The log info shown as below:
[rawvideo @ 0x35ea400] Estimating duration from bitrate, this may be inaccurate
Input #0, rawvideo, from '/root/media_streams//encoderbitstreams//1080p_blue_sky.yuv':
  Duration: 00:00:08.68, start: 0.000000, bitrate: 622080 kb/s
    Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p, 1920x1080, 622080 kb/s, 25 tbr, 25 tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> vp8 (vp8_vaapi))
Press [q] to stop, [?] for help
[vp8_vaapi @ 0x3610f40] Encoding entrypoint not found (14 / 6).
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
Conversion failed!

mypopydev, 2018-06-14:
which platform? in my SKL, use vainfo can't find the VP8 ENC entrypoint, so I think VP8 ENC fail in SKL is a expectable action if iHD didn't support VP8 ENC.

FocusLuo, 2018-06-21:
@XinfengZhang Carl, I used the CFL/KBL/GLK/SKL/APL platforms

XinfengZhang, 2018-06-21:
@focus, VP8  encode support is from CNL

XinfengZhang, 2018-07-09:
close it , for VP8 encode support is from CNL.

closed 2018-07-09 by XinfengZhang