intel/media-driver #195: Problem running on CFL

dbabokin, 2018-06-14:
I've installed Linux_C_for_Media_Developement_Package_20180413 package on Ubuntu 16.04.4 LTS (GNU/Linux 4.13.0-41-generic x86_64) running on i7-8700K (Coffee Lake). Trying to compile and run examples I get the following error:
libva info: VA-API version 1.0.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_0
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
CM-ERROR: CM_INVALID_LIBVA_INITIALIZE linear_walker.cpp 72

Is the problem due to unsupported hardware (KBL/CFL) or  this is the problem with driver installation / configuration?

dvrogozh, 2018-06-14:
Looks like you are running on 4.13 kernel.  I don't remember about KBL, but CFL support appeared only in 4.14 kernel. And if you run 4.14 you should start the kernel with `i915.alpha_support=1` since full support for CFL appeared only from 4.15.

On which HW your are running exactly? KBL or CFL?

As for the errors which you see. APERTURE error is actually a warning, this error is ignored - see from this line:
https://github.com/intel/media-driver/blob/master/media_driver/linux/common/os/libdrm/mos_bufmgr.c#L4976

So, this is a bad sign only and real error happens later on: chip identification failed. So, I think you need to update your kernel. I would suggest 4.16.

dbabokin, 2018-06-14:
I'm running on i7-8700K (Coffee Lake). I've upgraded to GNU/Linux 4.17.0-041700-generic x86_64. No effect. Though I might screw up somewhere in the upgrade process.

Are there any expected problems moving to Ubuntu 18.04 LTS? I might reinstalling OS if it may help..

dvrogozh, 2018-06-14:
> Are there any expected problems moving to Ubuntu 18.04 LTS?

I don't think there should be any problem, but I am not sure what is kernel version in this Ubuntu either.

> I've upgraded to GNU/Linux 4.17.0-041700-generic x86_64. No effect.

This should have worked. Have you rebooted:)? Well, what can still go wrong is that kernel mode driver is missing your CFL device id. Can you, please, provide the following:
1. Boot into 4.17, make sure you booted into it checking uname -a
2. Provide either full dmesg log from the start or at least output of the following command:
`# dmesg | grep drm`
3. Provide output of this command:
`# lspci -nn`
Let's check how kernel booted drm subsystem and what is your device ID.

dbabokin, 2018-06-14:
Installing 4.17 on 16.04 was a bad idea. Looks like it was not fulling installed. When I tried to downgrade I killed the machine. Now I need to reinstall and I will let you know the outcome.

HeFan2017, 2018-06-15:
hi @dbabokin , i can provide the steps to upgrade linux kernel 4.16 on Ubuntu16.04, where 4.16 supports CFL.
1. Download four packages from http://kernel.ubuntu.com/~kernel-ppa/mainline/v4.16.15/ 
a. linux-headers-4.16.15-041615_4.16.15-041615.201806111730_all.deb
b. linux-headers-4.16.15-041615-generic_4.16.15-041615.201806111730_amd64.deb
c. linux-modules-4.16.15-041615-generic_4.16.15-041615.201806111730_amd64.deb
d. linux-image-unsigned-4.16.15-041615-generic_4.16.15-041615.201806111730_amd64.deb
2. Install these packages one by one in order. Make sure every package is installed correctly. (when I installed them, it gave an error that libssl1.1 is not installed. Then just foundthe libssl1.1 package and installed it).
Please let me know it you have any questions.

dbabokin, 2018-06-22:
I've reinstalled the machine. Now I'm running Ubuntu 18.04 (Linux 4.15.0-23-generic). And I'm getting exactly the same error as above. Here's output of dmesg and lspci

> dmesg | grep drm
[    2.191289] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    2.191357] [drm] Driver supports precise vblank timestamp query.
[    2.632030] [drm] Cannot find any crtc or sizes
[    2.632121] [drm] Initialized nouveau 1.3.1 20120801 for 0000:01:00.0 on minor 0
[    2.824005] [drm] Cannot find any crtc or sizes
[    3.016005] [drm] Cannot find any crtc or sizes

> lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:3ec2] (rev 07)
00:01.0 PCI bridge [0604]: Intel Corporation Skylake PCIe Controller (x16) [8086:1901] (rev 07)
00:08.0 System peripheral [0880]: Intel Corporation Skylake Gaussian Mixture Model [8086:1911]
00:14.0 USB controller [0c03]: Intel Corporation 200 Series PCH USB 3.0 xHCI Controller [8086:a2af]
00:16.0 Communication controller [0780]: Intel Corporation 200 Series PCH CSME HECI #1 [8086:a2ba]
00:17.0 SATA controller [0106]: Intel Corporation 200 Series PCH SATA controller [AHCI mode] [8086:a282]
00:1b.0 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #17 [8086:a2e7] (rev f0)
00:1b.2 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #19 [8086:a2e9] (rev f0)
00:1b.4 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #21 [8086:a2eb] (rev f0)
00:1c.0 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #1 [8086:a290] (rev f0)
00:1c.2 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #3 [8086:a292] (rev f0)
00:1c.3 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #4 [8086:a293] (rev f0)
00:1c.4 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #5 [8086:a294] (rev f0)
00:1c.6 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #7 [8086:a296] (rev f0)
00:1d.0 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #9 [8086:a298] (rev f0)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a2c9]
00:1f.2 Memory controller [0580]: Intel Corporation 200 Series PCH PMC [8086:a2a1]
00:1f.3 Audio device [0403]: Intel Corporation 200 Series PCH HD Audio [8086:a2f0]
00:1f.4 SMBus [0c05]: Intel Corporation 200 Series PCH SMBus Controller [8086:a2a3]
00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (2) I219-V [8086:15b8]
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] [10de:1c03] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation GP106 High Definition Audio Controller [10de:10f1] (rev a1)
08:00.0 USB controller [0c03]: ASMedia Technology Inc. Device [1b21:2142]
09:00.0 Network controller [0280]: Intel Corporation Wireless 3165 [8086:3165] (rev 81)

According to reply above 4.15 should be good enough. Is it? Or should I upgrade to 4.16 or 4.17?

dvrogozh, 2018-06-22:
Both dmesg and lspci say that you have NVidea graphics card:
`[ 2.632121] [drm] Initialized nouveau 1.3.1 20120801 for 0000:01:00.0 on minor 0`
That's initialization of NVidea path in kernel mode driver.
`01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] [10de:1c03] (rev a1)`
That's self speaking...

And there is no mentioning of Intel Graphics which I can see. Maybe you simply have a system without it? Could you, please, provide the output of:
`less /proc/cpuinfo | grep "model name"`
Let's try to see description of your model on https://ark.intel.com.

dbabokin, 2018-06-22:
As I mentioned, I'm running on i7-8700K:
model name      : Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz
It has Intel® UHD Graphics 630.

Current software setup is mint Ubuntu 18.04 + several apt packages  + Intel drivers installation from "C for Media package" distribution. Nothing else. Do I need to install anything else?

uartie, 2018-06-22:
On dual GPU system, your nvidia GPU is likely being assigned to `/dev/dri/renderD128` and Intel GPU to `/dev/dri/renderD129` (when Intel GPU is enabled in BIOS).  Unfortunately, the media driver assumes `/dev/dri/renderD128` is Intel GPU and is hard-coded in the code.

You can try changing the code to use `/dev/dri/renderD129` and recompile (make sure Intel GPU is enabled in BIOS, too).

Perhaps someone should change the driver code to allow this to be configured at runtime.  And/Or add option in sample apps to allow the device to be configured.

See similar issue for intel-vaapi-driver (https://github.com/intel/intel-vaapi-driver/issues/186)

uartie, 2018-06-22:
Also, check `sudo cat /sys/kernel/debug/dri/128/name` or `sudo cat /sys/kernel/debug/dri/129/name` to see which one is using `i915` driver (i.e. Intel driver).

dvrogozh, 2018-06-24:
> Unfortunately, the media driver assumes /dev/dri/renderD128 is Intel GPU and is hard-coded in the code.

@uartie : this is wrong, or rather this is not exactly true :). Looking into the code:
1. media driver uses the drm FD which was passed by application. And that's totally up to application to use /dev/dri/card0 or /dev/dri/renderD128 or card1 and D129 in case of Intel card being the second. This is functional in case of DRM backend.
2. It looks like there is another path in driver initialization when drm FD was not passed. In this case driver tries to initialize drm FD on its own and here hardcoded /dev/dri/renderD128 appears: see the code starting fromhttps://github.com/intel/media-driver/blob/fc590d27d08ccf8f9a58c6defeb02230f6b3da5a/media_driver/linux/common/ddi/media_libva.cpp#L1170. I did not check, but probably we will step into this with non-DRM backends (X11 or Wayland)? However, I am not sure that this is a bug. If Intel Graphics is not the leading one, then why try to access it thru X11/Wayland? I would say that application needs to access it properly thru drm backend... Though I agree that as of now all that does not sound user-friendly... Need to think how to handle this situation...
3. 1) and 2) was about media-driver. But there is also CMRT. And here I am not sure what is going on. It is quite possible that this guy is _really_ hardcoding /dev/dri/renderD128 here https://github.com/intel/media-driver/blob/fc590d27d08ccf8f9a58c6defeb02230f6b3da5a/cmrtlib/linux/hardware/cm_device_os.cpp#L461

So, I think we need to have further discussion for 2) and 3).

> check sudo cat /sys/kernel/debug/dri/128/name or sudo cat /sys/kernel/debug/dri/129/name to see which one is using i915 driver (i.e. Intel driver).

More generally, please, provide output of these commands:
```
ls /dev/dri/
sudo cat /sys/kernel/debug/dri/128/name
sudo cat /sys/kernel/debug/dri/129/name
#or any other which you see as /dev/dri/renderDxxx
```
Unfortunately I did not work much with 2 graphics cards:(. But what bothers me is that I don't see i915 kmd being initialized by the kernel and lspci don't list Intel Graphics. I afraid that while your system looks indeed supporting Intel Graphics (https://ark.intel.com/products/126684/Intel-Core-i7-8700K-Processor-12M-Cache-up-to-4_70-GHz), you for some reason don't see it and can't access. I vaguely remember some BIOS settings which simply enable/disable graphics. Could that be that it is simply switched off there? If you will not see 2nd card in /dev/dri I would recommend to check BIOS settings...

dbabokin, 2018-06-26:
Indeed, I had iGPU switched off (shame on me). But after enabling it I see it mapped to renderD129 (render128 is NVidia). So now I'll need to recompile media-driver from sources.

So, would be really good to have initialization code to check the device and select appropriate one.

dvrogozh, 2018-06-26:
> Indeed, I had iGPU switched off (shame on me).

That's not trivial thing actually. I would step into the same for the first time.

> So now I'll need to recompile media-driver from sources.

Wait, wait, wait:). Why recompile?! As I said, drm backend should work... For example, see:
```
~/git/github/libva-utils/vainfo/vainfo -h
/home/dvrogozh/git/github/libva-utils/vainfo/vainfo: invalid option -- 'h'
Show information from VA-API driver
Usage: vainfo --help
        --help print this message

Usage: vainfo [options]
Display options:
        --display display | help         Show information for the specified display, or the available display list
        --device device                  Set device name, only available under drm display
```
Thus, first of all I would try:
`~/git/github/libva-utils/vainfo/vainfo --display drm --device /dev/dri/renderD129`

Next, what you are actually try to run? For example, ffmpeg-vaapi accepts device specification, for example:
```
ffmpeg \
   -hwaccel vaapi -vaapi_device /dev/dri/renderD129 -v verbose \
   -s:v 1920x1080 -i 1920x1080p.yuv -vf 'format=nv12,hwupload' -an \
   -c:v h264_vaapi -compression_level 4 -profile:v high -level:v 40 -g 50 -bf 2 -b:v 8M -maxrate:v 8M \
   -y vaapi.264
```
Should be able to run encoding via ffmpeg-vaapi on the Intel GPU.

dbabokin, 2018-06-26:
I'm trying to run CM examples, so I really need CMRT, not DRM backend. And after manually compiling media-driver (after fixing hardcoded /dev/dri/renderD128) I finally was able to successfully run the examples.

I'm ok closing this issue, but still would be good to (1) update the statement on KBL/CFL support on readme page (2) fix CMRT to use correct device in case of several GPUs in the system.

HeFan2017, 2018-06-26:
hi @dbabokin , we didn't consider the multi-GPU scenario before. We will work out a solution for that. Thanks for raising this issue.

XinfengZhang, 2019-08-27:
keep it open just because we need pay attention to multiple -GPU scenario

MartinX3, 2021-01-12:
Yes, same issue here breaking the `parole` media player from playing videos with gstreamer-backend

```
$ sudo cat /sys/kernel/debug/dri/128/name
nvidia-drm dev=0000:02:00.0 unique=0000:02:00.0
$ sudo cat /sys/kernel/debug/dri/129/name
i915 dev=0000:00:02.0 master=pci:0000:00:02.0 unique=0000:00:02.0
```

Now I need to find a way to force the i915 to be on dri/128 until this bug got fixed.

lorn10, 2022-04-19:
Also here, it looks that I have the same problem:
```
$ sudo cat /sys/kernel/debug/dri/128/name
radeon dev=0000:01:00.0 master=pci:0000:01:00.0 unique=0000:01:00.0
$ sudo cat /sys/kernel/debug/dri/129/name
i915 dev=0000:00:02.0 unique=0000:00:02.0
```
This is true for a multi-GPU system containing a **Radeon HD 6770M** and an **Intel HD 2000 iGPU**.

It would be really nice if that annoying problem could be fixed. Thanks! :+1:

XinfengZhang, 2022-04-20:
hi @lorn10 
1. you also need CMRT support?
2. does the method provided by MartinX3 works for you?

MartinX3, 2022-04-20:
> 2. does the method provided by MartinX3 works for you?

My solution is for Nvidia GPUs.

Isn't the amd driver part of the kernel and not a module?
Or do I misremember it?

lorn10, 2022-04-20:
I think one problem is in my case that the **Radeon dGPU** is the **primary GPU** (not the Intel one). Yes, this looks strange but this is the situation at the **Apple iMac 12,2** series computer (booted in native UEFI). Apple officially don't mention the iGPU, they had only advertised the Radeon dGPU. Nevertheless the Intel HD 2000 iGPU works under Linux, since the crocus driver is available I am able to start applications via the `DRI_PRIME=1` parameter, - even simply games like SuperTux are running fine. :wink: 

```
test@test-iMac:~$ sudo lshw -C display
  *-display                 
       description: VGA compatible controller
       product: Whistler [Radeon HD 6730M/6770M/7690M XT]
       vendor: Advanced Micro Devices, Inc. [AMD/ATI]
       physical id: 0
       bus info: pci@0000:01:00.0
       version: 00
       width: 64 bits
       clock: 33MHz
       capabilities: pm pciexpress msi vga_controller bus_master cap_list rom
       configuration: driver=radeon latency=0
       resources: irq:46 memory:90000000-9fffffff memory:a8800000-a881ffff ioport:2000(size=256) memory:a8820000-a883ffff
  *-display
       description: Display controller
       product: 2nd Generation Core Processor Family Integrated Graphics Controller
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       version: 09
       width: 64 bits
       clock: 33MHz
       capabilities: msi pm bus_master cap_list
       configuration: driver=i915 latency=0
       resources: irq:45 memory:a8000000-a83fffff memory:a0000000-a7ffffff ioport:3000(size=64)
```

MosesChuka, 2023-04-09:
**I am completely new to linux system.**

However, I was wondering if there was a fix for this issue on Parallel Desktop based linux.

running the `vainfo` command gives the following:
<sub>
mcmoses@ubuntu:/$ vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'i965'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_19
DRM_IOCTL_I915_GEM_APERTURE failed: Bad file descriptor
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [9]
param: 4, val: 0
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
</sub>

> `sudo cat /sys/kernel/debug/dri/128/name` gives:
<sub>prl_vid dev=0000:01:00.0 master=pci:0000:01:00.0 unique=0000:01:00.0</sub>

>and `sudo cat /sys/kernel/debug/dri/129/name` gives:
<sub>cat: /sys/kernel/debug/dri/129/name: No such file or directory</sub>

**Background**
Am try to run the `./bin/Protonect` for Kinectv2