intel/media-driver #196: [ffmpeg-vaapi][mjpeg decode] mjpeg decode test cases failed when using ffmpeg-vaapi

FocusLuo, 2018-06-15:
These test cases can work on intel-vaapi-driver, but failed on intel-media-driver
One CMD line shown as below:
ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -v verbose -i /media//mjpeg/MOBILE_352x288_30_orig_01.mjpeg -pix_fmt yuv420p -vframes 50 -y ./dump.yuv

[mjpeg @ 0x3ec3500] Format mjpeg detected only with low score of 25, misdetection possible!
Input #0, mjpeg, from './MOBILE_352x288_30_orig_01.mjpeg':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, 1 reference frame, yuvj420p(pc, bt470bg/unknown/unknown, center), 352x288 [SAR 1:1 DAR 11:9], 25 tbr, 1200k tbn, 25 tbc
[AVHWDeviceContext @ 0x3ec66c0] Opened VA display via DRM device /dev/dri/renderD128.
[AVHWDeviceContext @ 0x3ec66c0] libva: VA-API version 1.1.0
[AVHWDeviceContext @ 0x3ec66c0] libva: va_getDriverName() returns 0
[AVHWDeviceContext @ 0x3ec66c0] libva: User requested driver 'iHD'
[AVHWDeviceContext @ 0x3ec66c0] libva: Trying to open /opt/X11R7/vpg_driver/lib/dri//iHD_drv_video.so
[AVHWDeviceContext @ 0x3ec66c0] libva: Found init function __vaDriverInit_1_1
[AVHWDeviceContext @ 0x3ec66c0] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x3ec66c0] Initialised VAAPI connection: version 1.1
[AVHWDeviceContext @ 0x3ec66c0] Unknown driver "Intel iHD driver - 2.0.0", assuming standard behaviour.
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> rawvideo (native))
Press [q] to stop, [?] for help
[AVHWFramesContext @ 0x3f2d640] Failed to read image from surface 0x5: 20 (the requested function is not implemented).
[mjpeg @ 0x3ec52c0] Failed to transfer data to output frame: -5.
Error while processing the decoded data for stream #0:0
Conversion failed!

FocusLuo, 2018-06-15:
attached the test file used by the test cmd line
[MOBILE_352x288_30_orig_01.zip](https://github.com/intel/media-driver/files/2105401/MOBILE_352x288_30_orig_01.zip)

mypopydev, 2018-06-17:
I think the root cause is iHD can't dump the raw YUV data in JPEG dec, and I guess iHD driver need to double-check vaCreateImage/vaGetImage

dvrogozh, 2018-06-17:
>  guess iHD driver need to double-check vaCreateImage/vaGetImage

Quite possible. iHD driver was paired with mediasdk for the very long time and mediasdk don't use that uAPI. As a result you can expect malfunction support.

XinfengZhang, 2018-06-19:
I cant duplicate the issue with commit #97916802ee0964b4a2b03237ce0606dd2b35351f
result YUV is fine. @FocusLuo please help to check it.
 ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -v verbose -i ~/Downloads/MOBILE_352x288_30_orig_01.mjpeg -pix_fmt yuv420p -vframes 50 -y ./dump.yuv
ffmpeg version N-89830-g0afa171 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 4.8.5 (GCC) 20150623 (Red Hat 4.8.5-4)
  configuration:
  libavutil      56.  7.100 / 56.  7.100
  libavcodec     58.  9.100 / 58.  9.100
  libavformat    58.  5.100 / 58.  5.100
  libavdevice    58.  0.100 / 58.  0.100
  libavfilter     7. 11.101 /  7. 11.101
  libswscale      5.  0.101 /  5.  0.101
  libswresample   3.  0.101 /  3.  0.101
[mjpeg @ 0x367f500] Format mjpeg detected only with low score of 25, misdetection possible!
Input #0, mjpeg, from '/home/intel/Downloads/MOBILE_352x288_30_orig_01.mjpeg':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, 1 reference frame, yuvj420p(pc, bt470bg/unknown/unknown, center), 352x288 [SAR 1:1 DAR 11:9], 25 tbr, 1200k tbn, 25 tbc
[AVHWDeviceContext @ 0x3682a00] Opened VA display via DRM device /dev/dri/renderD128.
[AVHWDeviceContext @ 0x3682a00] libva: VA-API version 1.1.0
[AVHWDeviceContext @ 0x3682a00] libva: va_getDriverName() returns 0
[AVHWDeviceContext @ 0x3682a00] libva: User requested driver 'iHD'
[AVHWDeviceContext @ 0x3682a00] libva: Trying to open /opt/intel/mediasdk/lib64/iHD_drv_video.so
[AVHWDeviceContext @ 0x3682a00] libva: Found init function __vaDriverInit_1_1
[AVHWDeviceContext @ 0x3682a00] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x3682a00] Initialised VAAPI connection: version 1.1
[AVHWDeviceContext @ 0x3682a00] Unknown driver "Intel iHD driver - 2.0.0", assuming standard behaviour.
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> rawvideo (native))
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x3710240] w:352 h:288 pixfmt:yuvj420p tb:1/1200000 fr:25/1 sar:1/1 sws_param:flags=2
[auto_scaler_0 @ 0x3710140] w:iw h:ih flags:'bicubic' interl:0
[format @ 0x3711600] auto-inserting filter 'auto_scaler_0' between the filter 'Parsed_null_0' and the filter 'format'
[swscaler @ 0x37120c0] deprecated pixel format used, make sure you did set range correctly
[auto_scaler_0 @ 0x3710140] w:352 h:288 fmt:yuvj420p sar:1/1 -> w:352 h:288 fmt:yuv420p sar:1/1 flags:0x4
Output #0, rawvideo, to './dump.yuv':
  Metadata:
    encoder         : Lavf58.5.100
    Stream #0:0: Video: rawvideo, 1 reference frame (I420 / 0x30323449), yuv420p(center), 352x288 [SAR 1:1 DAR 11:9], q=2-31, 30412 kb/s, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.9.100 rawvideo
No more output streams to write to, finishing.
frame=   50 fps=0.0 q=-0.0 Lsize=    7425kB time=00:00:02.00 bitrate=30412.8kbits/s speed=43.8x
video:7425kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%
Input file #0 (/home/intel/Downloads/MOBILE_352x288_30_orig_01.mjpeg):
  Input stream #0:0 (video): 50 packets read (2092389 bytes); 50 frames decoded;
  Total: 50 packets (2092389 bytes) demuxed
Output file #0 (./dump.yuv):
  Output stream #0:0 (video): 50 frames encoded; 50 packets muxed (7603200 bytes);
  Total: 50 packets (7603200 bytes) muxed

uartie, 2018-06-19:
@XinfengZhang looks like you have ffmpeg from 2018-01-19 commit... can you check with latest ffmpeg master?

mypopydev, 2018-06-20:
I can reproduce with FFmpeg master/iHD master branch

XinfengZhang, 2018-06-21:
sure, I will update my ffmpeg to check again

XinfengZhang, 2018-07-09:
yes, the surface format is IMC3, but the image format is YV12,  driver does not call vp inside vaGetImage