intel/media-driver #197: build failure with particular cmake c_flags system configuration

chivakker, 2018-06-15:
Using https://github.com/intel/media-driver/commit/97916802ee0964b4a2b03237ce0606dd2b35351f.

The following building error is seen with a cmake system configuration as follows.  Overriding CMAKE_CXX_FLAGS="" will make the compilation go through.   I cannot change my system configuration so please use it to trigger the issue.  

These are the CMAKE_CXX_FLAGS ( taken from output of cmake --system-configuration)

CMAKE_CXX_FLAGS "-g -O3 -feliminate-unused-debug-types  -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=32 -Wformat -Wformat-security -Wl,--copy-dt-needed-entries -m64  -fasynchronous-unwind-tables -Wp,-D_REENTRANT -ftree-loop-distribute-patterns -Wl,-z -Wl,now -Wl,-z -Wl,relro -fno-semantic-interposition -ffat-lto-objects  -fno-signed-zeros -fno-trapping-math  -fassociative-math -Wl,-sort-common -Wl,--enable-new-dtags -fvisibility-inlines-hidden -Wl,--enable-new-dtags"

following the guide to build media-driver, this is a snip of errors I'm seeing 

media-driver/cmrtlib/agnostic/share/cm_printf_host.cpp:629:63: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
                     m_args[m_numMultArg - m_argsExpected] = *((int*            )&(header->scalar64));
                                                              ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
media-driver/cmrtlib/agnostic/share/cm_printf_host.cpp:632:63: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
                     m_args[m_numMultArg - m_argsExpected] = *((unsigned int*   )&(header->scalar64));
                                                              ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
media-driver/cmrtlib/agnostic/share/cm_printf_host.cpp:641:63: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
                     m_args[m_numMultArg - m_argsExpected] = *((short*         )&(header->scalar64));

oliver-sang, 2018-06-20:
whether this can be fixed by https://github.com/intel/media-driver/pull/199 ?

jwang11, 2018-06-20:
@oliver-sang I think yes

oliver-sang, 2018-06-27:
@chivakker can you check whether it fixed since we've already merged #199 ? Thanks

chivakker, 2018-06-27:
yes, it fixed the problem for Ubuntu and Fedora.  There's another problem with ULT and pciaccess that we're trying to address on other bug #213

oliver-sang, 2018-06-28:
ok, since #213 is used to track specific problem, let me close this one.

closed 2018-06-28 by oliver-sang