intel/media-driver #198: provide a default building configuration

chivakker, 2018-06-15:
Please provide a default building configuration where the following compiles something that can be used.

workdir/media-driver
workdir/gmmlib

$ cd workdir/build_media
$ cmake -DCMAKE_INSTALL_PREFIX=<my own prefix>
$ make install

This is something MediaSDK project implemented and is really helpful.

oliver-sang, 2018-06-20:
ok, will work on this.
For CMAKE_INSTALL_PREFIX=, there are actually two libraries in media-driver to compile - iHD (media driver) and cmrtlib driver. iHD is the backend of libva, so in current cmake, the iHD install path is determined by libva config by default. cmrtlib is just a normal library to install to system path. So what's your suggestion how CMAKE_INSTALL_PREFIX controls the install behavior for both iHD and cmrtlib? Thanks

oliver-sang, 2018-06-27:
@chivakker how about cmrtlib installed directly into CMAKE_INSTALL_PREFIX and iHD installed into CMAKE_INSTALL_PREFIX/dri ?

chivakker, 2018-06-27:
@oliver-sang yeah that's the default being used with the drivers, it should be ok.  I am also wondering why we need cmrtlib compiled and installed by default if there's no opencl use cases needed for iHD to operate.  Maybe you want to attack this problem later but I thought it was worth mentioning.

dvrogozh, 2018-06-27:
> I am also wondering why we need cmrtlib compiled and installed by default if there's no opencl use cases needed for iHD to operate

cmrtlib is not related to opencl. For example, mediasdk uses it directly to improve performance or quality in a certain use cases.

Now few speculations:)...
I think cmrt sources are coupled with the driver simply because driver is implementation is based on it, I mean that driver don't use cmrtlib, but it uses core features (cm) which cmrtlib exposes. I.e. sources are in the driver repository, cmrtlib simply exposes them for users. I guess that if we will want to move cmrt out to its own repo we will simply end up duplicating the media driver... Though, it should be possible to make a more deep refactoring and extract out cm completely transforming it to the real driver dependency... Hm.... Need to discuss this with @XinfengZhang and @oliver-sang... Well, I would like to hear from these 2 guys confirmation on what I just wrote... In intel-vaapi-driver you also had this part actually which was a shaders layer and you used some asm-4gen or something similar to compile the asm shaders code - this is analogue of cmrt, though some principals are different...

oliver-sang, 2018-06-28:
@HeFan2017 could you help comment?

HeFan2017, 2018-06-28:
hi @dvrogozh , you are all right about the cmrtlib. And another point it that we can now hardly extract out cm completely because cm is not a pure dependency of media driver, it also relies on the bottom layer of media driver (renderhal/mhw/mos).

PR #223 (2018-08-08): Resolve 'provide a default building configuration'
Fixes: #198  
 - Default CMAKE_INSTALL_PREFIX to '/usr/local' but allow modifications at run-time
 - Default MEDIA_VERSION / UFO_VERSION to 1.0.0
 - Default gmmlib dependency paths until builds shift to dynamic linked
 - Update README.md to reflect simplified build steps
 - Update distro_cmake.sh & unified_cmake.sh scripts

Change-Id: I9250f6ff07c670a6cdf37ea516513b031508cabb
commit 54ca9bed (2018-08-08): Resolve 'provide a default building configuration'
 Fixes: #198
 - Default CMAKE_INSTALL_PREFIX to '/usr/local' if empty string
 - Default MEDIA_VERSION / UFO_VERSION to 1.0.0
 - Default gmmlib dependency paths until builds shift to dynamic linked
 - Update README.md to reflect simplified build steps
 - Update distro_cmake.sh & unified_cmake.sh scripts

Change-Id: I9250f6ff07c670a6cdf37ea516513b031508cabb
closed 2018-08-08 by intel-mediadev