intel/media-driver #200: [SKL] dump a lot of "drmIoctl:DRM_IOCTL_I915_GETPARAM with unsupport type"

mypopydev, 2018-06-20:
Build master branch with:
cmake ../media-driver -DCMAKE_INSTALL_PREFIX=/usr -DMEDIA_VERSION="2.0.3" -DBS_DIR_GMMLIB=`pwd`/../gmmlib/Source/GmmLib/ -DBS_DIR_COMMON=`pwd`/../gmmlib/Source/Common/ -DBS_DIR_INC=`pwd`/../gmmlib/Source/inc/ -DBS_DIR_MEDIA=`pwd`/../media-driver -DBUILD_TYPE=Debug

barry@barry-HP-ENVY-Notebook:~/Sources/umd_dev/build$ uname -a
Linux barry-HP-ENVY-Notebook 4.15.0-23-generic #25-Ubuntu SMP Wed May 23 18:02:16 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
barry@barry-HP-ENVY-Notebook:~/Sources/umd_dev/build$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04 LTS
Release:	18.04
Codename:	bionic

then after make install will get a lot of "drmIoctl:DRM_IOCTL_I915_GETPARAM with unsupport type"

oliver-sang, 2018-06-22:
Thanks for reporting. Under investigation now.

oliver-sang, 2018-06-29:
@mypopydev have your problem fixed by 43f881bc64b62aac0dd9f4d9408f59aaf5d20aa5 ? Thanks

mypopydev, 2018-06-30:
FIxed, thanks @oliver-sang

closed 2018-06-30 by oliver-sang