intel/media-driver #201: Driver doesn't report VPP maxWidth/maxHeight via vaQueryVideoProcPipelineCaps

dmitryermilov, 2018-06-20:
Driver doesn't report VPP maxWidth/maxHeight via vaQueryVideoProcPipelineCaps for EUs (e.g. scaling) and VEBOX usage cases. See code below:
https://github.com/intel/media-driver/blob/master/media_driver/linux/common/ddi/media_libva.cpp#L5343-L5363

The driver fills width/height/pixel_format caps **only** for SFC cases.

XinfengZhang, 2018-06-22:
thanks for reporting, we will add the missing part

dmitryermilov, 2018-07-11:
Hi @XinfengZhang . Is there any progress ?

FurongZhang, 2018-07-29:
@dmitryermilov We have discussed this issue. And it should be in open source very soon. I will check the code if it is there now. Thanks for your feedback.

dvrogozh, 2018-08-01:
Folks, please, stop pushing changes to repository without review! This should be a PR first, which we need to take a look, try and comment. For example, in the above commit I left 2 comments:
1. Shouldn't that be a per-platform specific?
2. Shouldn't that be a SFC/AVS specific?

PR #263 (2018-08-30): [VP]Add platform information in DdiMedia_QueryVideoProcPipelineCaps
Fix #201, add platform information in DdiMedia_QueryVideoProcPipelineCaps to distinguish Gen8 and Gen9+ max width and height
closed 2018-08-30 by Sherry-Lin