intel/media-driver #204: [ffmpeg-vaapi][hevc 10bit decode] hevc 10bit decode test cases failed when using ffmpeg-vaapi

FocusLuo, 2018-06-21:
The CMD line can work on i965 driver, but failed(core dumped) on iHD driver
 ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -v verbose -y -i ./WPP_A_ericsson_MAIN10_2.bin -pix_fmt p010le -vframes 25 -f md5 vaapi_out.md5

[AVHWDeviceContext @ 0x561e7c139500] Opened VA display via DRM device /dev/dri/renderD128.
[AVHWDeviceContext @ 0x561e7c139500] libva: VA-API version 1.1.0
[AVHWDeviceContext @ 0x561e7c139500] libva: va_getDriverName() returns 0
[AVHWDeviceContext @ 0x561e7c139500] libva: User requested driver 'iHD'
[AVHWDeviceContext @ 0x561e7c139500] libva: Trying to open /opt/X11R7/vpg_driver/lib/dri//iHD_drv_video.so
[AVHWDeviceContext @ 0x561e7c139500] libva: Found init function __vaDriverInit_1_1
[AVHWDeviceContext @ 0x561e7c139500] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x561e7c139500] Initialised VAAPI connection: version 1.1
[AVHWDeviceContext @ 0x561e7c139500] Unknown driver "Intel iHD driver - 2.0.0", assuming standard behaviour.
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (native) -> rawvideo (native))
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x561e7c1bbdc0] w:416 h:240 pixfmt:p010le tb:1/1200000 fr:25/1 sar:0/1 sws_param:flags=2
Output #0, md5, to 'vaapi_out.md5':
  Metadata:
    encoder         : Lavf58.17.100
    Stream #0:0: Video: rawvideo, 1 reference frame, p010le, 416x240, q=2-31, 37440 kb/s, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.20.101 rawvideo
Segmentation fault (core dumped)

XinfengZhang, 2018-06-26:
@FocusLuo please help to add platform information

XinfengZhang, 2018-06-28:
root caused. vaCreateSurface will use gmmlib to manager allocation size for a P010 surface (416x240 to 896x256), but vaCreateImage does not use gmmlib, so the size is mismatch when call vaGetSurface. will fix it ASAP

zhuqingliang, 2018-08-16:
This issues still can be reproduced with media_driver  36b3a0aa3b47a4cda36a31d98e16da9ff7ea27fb .
HEVC_10bit_allegro10 : passed 173 failed 42 on CFL , passed 175 failed 40 on KBL .
hevc_10bit_jvt              : only TSUNEQBD_A_MAIN10_Technicolor_2.bin failed.

XinfengZhang, 2019-08-27:
@zhuqingliang ,  I will close the issue, some cases pass, some cases failed, we need a new issue which can be reproduced easily to handle it

closed 2019-08-27 by XinfengZhang