intel/media-driver #207: [ffmpeg-vaapi][HEVC encode] multi-process encode failed by using ffmpeg-vaapi

FocusLuo, 2018-06-21:
To start 8 process for a same hevc encode comline, and then to check the md5 for each output file, the iHD driver have different md5, but the i965 driver can work well
for ((i=0;i<8;i++));do
{
     ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -s:v 720x480 -i ./Test_720x480p.yuv -an -vf format=nv12,hwupload -c:v hevc_vaapi -g 10 -bf 1 -refs 2 -vframes 100 -y ./Hevce_multi_process_720x480P_"$i".h265
} &
done
wait $!

XinfengZhang, 2018-06-26:
cant reproduce with latest ffmpeg (4ac88ba5487e026bf81da565f97cfcf8f920657d)

wangzj0601, 2018-06-29:
this issue can be duplicated on CFL/KBL/GLK, while can not be duplicated on SKL with below env.
media-driver: b06a5736095c2b9c735ea42a64de9c7f3a9191bb
ffmpeg: 27662ed681cbc74e9492cecf86ead8946679fd23 repo:https://git.ffmpeg.org/ffmpeg.git

XinfengZhang, 2018-07-08:
@wangzj0601 suppose even 1 session cant work on CFL/KBL, because lack of P frame support? right?

zhuqingliang, 2018-11-12:
passed at latest codebase ffmpeg(n4.1)