intel/media-driver #208: [ffmpeg-vaapi][vc1 decode] most of VC1 decode test cases failed by using ffmpeg-vaapi

FocusLuo, 2018-06-21:
One CMD line as below, which can work well on i965 driver
ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -v verbose -i ./SA10110.vc1 -pix_fmt yuv420p -y ./SA10110.vc1.yuv

[auto_scaler_0 @ 0x5624b34fb680] w:720 h:480 fmt:nv12 sar:1/1 -> w:720 h:480 fmt:yuv420p sar:1/1 flags:0x4
Output #0, rawvideo, to './SA10110.vc1.yuv':
  Metadata:
    encoder         : Lavf58.17.100
    Stream #0:0: Video: rawvideo, 1 reference frame (I420 / 0x30323449), yuv420p(left), 720x480 [SAR 1:1 DAR 3:2], q=2-31, 103680 kb/s, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.20.101 rawvideo
*** 1 dup!
[AVHWFramesContext @ 0x5624b344ebc0] Failed to sync surface 0x4: 23 (internal decoding error).
[vc1 @ 0x5624b337ddc0] Failed to transfer data to output frame: -5.
Error while processing the decoded data for stream #0:0

XinfengZhang, 2018-08-14:
#266 should fix it

zhuqingliang, 2018-08-15:
There is still only one case SA30031.vc1 failed with media_driver commit 36b3a0aa3b47a4cda36a31d98e16da9ff7ea27fb now, and the dumped file has garbage in every frame.

Tianhaol, 2018-09-05:
The output yuv file can be displayed correctly by our internal tool.

FocusLuo, 2018-10-13:
Closed this bug for almost all test cases passed

closed 2018-10-13 by FocusLuo