intel/media-driver #209: [ffmpeg-vaapi][HEVC CQP encode] the cqp encode failed on CFL/KBL platforms by using ffmpeg-vaapi

FocusLuo, 2018-06-21:
the hevc cqp encode fail on CFL and KBL platforms, but can work on SKL,GLK,APL paltforms
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -s:v 3840x2160 -i ./4K_2013_3840_2160_300frames.yuv -vf format=nv12,hwupload -c:v hevc_vaapi -g 12 -bf 0 -qp 36 -quality 4 -vframes 100 -y ./hevce_cqp_3840x2160_gop12_refdist0_qp36.h265
[hevc_vaapi @ 0x55865cf8f440] Failed to end picture encode issue: 24 (internal encoding error).
[hevc_vaapi @ 0x55865cf8f440] Encode failed: -5.
Video encoding failed

Brainiarc7, 2018-06-27:
Hello there,

Neither Kabylake (KBL) nor Coffeelake (CFL) are supported at the moment by this driver.

If its' VAAPI you're after, switch to the intel-vaapi-driver here: https://github.com/intel/intel-vaapi-driver

Brainiarc7, 2018-06-27:
See https://github.com/intel/media-driver/blob/master/README.md for supported platforms.

XinfengZhang, 2018-07-08:
related with P frame support

XinfengZhang, 2018-11-19:
P frame supported.

XinfengZhang, 2018-11-19:
from 5c8a663

zhuqingliang, 2018-11-20:
some other case of cqp encode failed:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -s:v 352x288 -i ./MOBILE_352x288_30_orig_01.yuv -vf format=nv12,hwupload -c:v hevc_vaapi -g 30 -bf 3 -qp 46 -quality 4 -vframes 100 -y ./hevce_cqp_352x288_gop30_refdist3_qp46.h265