intel/media-driver #213: Build error: libdrm_mock.so: undefined symbol: pci_system_cleanup

asifhg, 2018-06-25:
NOTE:
1) This is not a duplicate of issue #127. However, the failure may be the same as in issue #127.
2) This failure is not reproducible for source repo at commit ID:
   923731f2e8 [Encode] Fix IFrameDist surface programming issue(CL#776073).
2.1) The cmake cmdline used to make it work:
     cmake ../${p} \
        -DCMAKE_INSTALL_PREFIX=$MY_INSTALL_DIR \
        -DMEDIA_VERSION=2.0.0 \
        -DBS_DIR_GMMLIB=[ pwd ]/../gmmlib/Source/GmmLib/ \
        -DBS_DIR_COMMON=[ pwd ]/../gmmlib/Source/Common/ \
        -DBS_DIR_INC=[ pwd ]/../gmmlib/Source/inc/ \
        -DBS_DIR_MEDIA=[ pwd ]/../$MY_INSTALL_DIR \
        **-DCMAKE_CXX_FLAGS=""** ;
2.2) Failure occurs even though, -fno-plt compiler flag is not used.

**Compile-time Failure Trigger**
**----------------------------**
LD_PRELOAD=../libdrm_mock/libdrm_mock.so ./devult ../../../iHD_drv_video.so ;

**Compile-time Failure Message**
**----------------------------**
symbol lookup error: ../libdrm_mock/libdrm_mock.so: undefined symbol: pci_system_cleanup

**ldd libdrm_mock.so**
**------------------**
ldd build_media/media_driver/linux/ult/libdrm_mock/libdrm_mock.so ;
	linux-vdso.so.1 (0x00007ffcc0b9e000)
	libstdc++.so.6 => /usr/lib64/haswell/libstdc++.so.6 (0x00007fc88a2b0000)
	libm.so.6 => /usr/lib64/haswell/libm.so.6 (0x00007fc88a12e000)
	libgcc_s.so.1 => /usr/lib64/haswell/libgcc_s.so.1 (0x00007fc88a114000)
	libc.so.6 => /usr/lib64/haswell/libc.so.6 (0x00007fc889f2c000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007fc88a893000)

**Compiler Options**
**----------------**
OS = ClearLinux 23240
gcc version = gcc (Clear Linux OS for Intel Architecture) 8.1.1 20180514
pkg-config version = 0.29.2

**Additional gcc compiler flags used compared to passing case on Ubuntu 16.04 = \**
   --param=ssp-buffer-size=32
   -O3
   -Wformat
   -Wformat-security
   -Wl,--copy-dt-needed-entries
   -Wl,--enable-new-dtags
   -Wl,-sort-common
   -Wl,-z
   -Wl,now
   -Wl,relro
   -Wp,-D_FORTIFY_SOURCE=2
   -Wp,-D_REENTRANT
   -fassociative-math
   -fasynchronous-unwind-tables
   -feliminate-unused-debug-types
   -ffat-lto-objects
   -fno-semantic-interposition
   -fno-signed-zeros
   -fno-trapping-math
   -ftree-loop-distribute-patterns
   -pipe

**Additional gcc linker flags used compared to passing case on Ubuntu 16.04 = \**
   --param=ssp-buffer-size=32
   -Wno-reorder
   -O3
   -Wall
   -Wformat
   -Wformat-security
   -Wl,--copy-dt-needed-entries
   -Wl,--enable-new-dtags
   -Wl,-sort-common
   -Wl,-z
   -Wl,now
   -Wl,relro
   -Wp,-D_FORTIFY_SOURCE=2
   -Wp,-D_REENTRANT
   -fassociative-math
   -fasynchronous-unwind-tables
   -feliminate-unused-debug-types
   -ffat-lto-objects
   -fno-semantic-interposition
   -fno-signed-zeros
   -fno-trapping-math
   -fstack-protector
   -ftree-loop-distribute-patterns
   -g
   -m64
   -pipe

oliver-sang, 2018-06-26:
Thanks for reporting. under investigation now.

walter-bai, 2018-06-26:
pci_system_cleanup along with a bunch of some other functions are defined in libpciaccess and neither libdrm_mock nor devult links this lib.                                    
  I checked on Ubuntu that libpciaccess is in /usr/lib/x86_64-linux-gnu/ that will be loaded by the kernel at startup, so this issue does not reproduce on Ubuntu. I don’t know how Clear Linux load libpciaccess, maybe that’s the cause.
  I also found that in recent version of driver, pci_system_cleanup is not called, so issue does not reproduce.

asifhg, 2018-06-26:
FYI, on my Ubuntu 16.04 system these are the libs in which I find pci_system_cleanup:
**/usr/lib/x86_64-linux-gnu/libpciaccess.a**
0000000000000020 g     F .text	0000000000000138 pci_system_cleanup
**/usr/lib/x86_64-linux-gnu/libdrm_intel.a**
0000000000000000         *UND*	0000000000000000 pci_system_cleanup

On my ClearLinux system, however, I am not able to find any libraries which contain pci_system_cleanup!

walter-bai, 2018-06-27:
Since libpciaccess is a driver dependency, it should be installed first.

uartie, 2018-06-27:
If libpciaccess is a dependency, then cmake should check for it during configuration stage... which would fail with more user-friendly message if it's missing (and a hint to user on how to satisfy the requirement).  As a general rule/practice, all dependencies should be checked during configuration stage so end-users don't encounter these types of cryptic compiler errors and consume everyone's time reporting and debugging.

I honestly hope this is patched in cmake (as aforementioned), instead of considered resolved by previous comment.

walter-bai, 2018-06-27:
That's a good idea, will fix it later.

jwang11, 2018-07-02:
On latest ClearLinux, I can see pci_system_clearup by
```
$objdump -T /usr/lib64/libpciaccess.so|grep pci_system_cleanup
0000000000003d80 g    DF .text  0000000000000149  Base        pci_system_cleanup
```
From my observation, the problem actually lie in media_driver/linux/ult/libdrm_mock/CMakeLists.txt, which lack of pciaccess link dependency. After append below lines, the issue disappear.
```
add_library(drm_mock SHARED ${SOURCES})
+find_library(PCIACCESS_LIBRARIES pciaccess)
+target_link_libraries(drm_mock ${PCIACCESS_LIBRARIES})
```

walter-bai, 2018-07-02:
Thanks for your comment. In a recent change 73313af0f64c75576ed18d406c88ac4cbdec2bf4, I have linked pciaccess  with drm_mock.

oliver-sang, 2018-07-11:
@asifhg could please check whether your problem has been solved in latest code? Thanks

oliver-sang, 2018-07-25:
didn't get further complain about this, close this issue. If found the further problem, please submit new issue. Thanks

closed 2018-07-25 by oliver-sang