intel/media-driver #214: Cannot build the driver on Ubuntu 18.04LTS

Brainiarc7, 2018-06-27:
Hello there,

This is the error I run into when building the intel-media-driver package as per the build instructions:

```
In file included from /home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.h:30:0,
                 from /home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:27:
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva.h:213:5: error: ‘VASliceParameterBufferHEVCExtension’ does not name a type; did you mean ‘VASliceParameterBufferHEVC’?
     VASliceParameterBufferHEVCExtension          *pVASliceParaBufHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VASliceParameterBufferHEVC
[ 79%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_mpeg2.cpp.o
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva.h:219:5: error: ‘VAPictureParameterBufferHEVCExtension’ does not name a type; did you mean ‘VAPictureParameterBufferHEVC’?
     VAPictureParameterBufferHEVCExtension        PicParamHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VAPictureParameterBufferHEVC
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:8270: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_base.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_base.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_ddi_base.h:34:0,
                 from /home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.h:31,
                 from /home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:29:
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva.h:213:5: error: ‘VASliceParameterBufferHEVCExtension’ does not name a type; did you mean ‘VASliceParameterBufferHEVC’?
     VASliceParameterBufferHEVCExtension          *pVASliceParaBufHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VASliceParameterBufferHEVC
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva.h:219:5: error: ‘VAPictureParameterBufferHEVCExtension’ does not name a type; did you mean ‘VAPictureParameterBufferHEVC’?
     VAPictureParameterBufferHEVCExtension        PicParamHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VAPictureParameterBufferHEVC
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp: In member function ‘virtual VAStatus DdiEncodeBase::StatusReport(DDI_MEDIA_BUFFER*, void**)’:
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:204:20: error: ‘VA_STATUS_ERROR_NOT_ENOUGH_BUFFER’ was not declared in this scope
             return VA_STATUS_ERROR_NOT_ENOUGH_BUFFER;
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:204:20: note: suggested alternative: ‘MOS_STATUS_NOT_ENOUGH_BUFFER’
             return VA_STATUS_ERROR_NOT_ENOUGH_BUFFER;
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                    MOS_STATUS_NOT_ENOUGH_BUFFER
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:8294: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_encode_base.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_encode_base.cpp.o] Error 1
In file included from /home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.h:30:0,
                 from /home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:33:
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva.h:213:5: error: ‘VASliceParameterBufferHEVCExtension’ does not name a type; did you mean ‘VASliceParameterBufferHEVC’?
     VASliceParameterBufferHEVCExtension          *pVASliceParaBufHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VASliceParameterBufferHEVC
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva.h:219:5: error: ‘VAPictureParameterBufferHEVCExtension’ does not name a type; did you mean ‘VAPictureParameterBufferHEVC’?
     VAPictureParameterBufferHEVCExtension        PicParamHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VAPictureParameterBufferHEVC
In file included from /home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_libva_encoder.h:30:0,
                 from /home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_libva_encoder.cpp:27:
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva.h:213:5: error: ‘VASliceParameterBufferHEVCExtension’ does not name a type; did you mean ‘VASliceParameterBufferHEVC’?
     VASliceParameterBufferHEVCExtension          *pVASliceParaBufHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VASliceParameterBufferHEVC
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva.h:219:5: error: ‘VAPictureParameterBufferHEVCExtension’ does not name a type; did you mean ‘VAPictureParameterBufferHEVC’?
     VAPictureParameterBufferHEVCExtension        PicParamHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VAPictureParameterBufferHEVC
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:8318: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_libva_decoder.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_libva_decoder.cpp.o] Error 1
In file included from /home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.h:30:0,
                 from /home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_avc.cpp:27:
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva.h:213:5: error: ‘VASliceParameterBufferHEVCExtension’ does not name a type; did you mean ‘VASliceParameterBufferHEVC’?
     VASliceParameterBufferHEVCExtension          *pVASliceParaBufHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VASliceParameterBufferHEVC
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:8342: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_libva_encoder.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_libva_encoder.cpp.o] Error 1
[... 23 lines cut ...]
     VAPictureParameterBufferHEVCExtension        PicParamHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VAPictureParameterBufferHEVC
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva.h:219:5: error: ‘VAPictureParameterBufferHEVCExtension’ does not name a type; did you mean ‘VAPictureParameterBufferHEVC’?
     VAPictureParameterBufferHEVCExtension        PicParamHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VAPictureParameterBufferHEVC
/home/brainiarc7/vaapi/workspace/media-driver/media_driver/linux/common/ddi/media_libva.h:219:5: error: ‘VAPictureParameterBufferHEVCExtension’ does not name a type; did you mean ‘VAPictureParameterBufferHEVC’?
     VAPictureParameterBufferHEVCExtension        PicParamHEVCRext;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     VAPictureParameterBufferHEVC
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:8366: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_avc.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_avc.cpp.o] Error 1
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:8390: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_hevc.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_hevc.cpp.o] Error 1
[... cut ...]
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:8438: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_mpeg2.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_mpeg2.cpp.o] Error 1
CMakeFiles/Makefile2:167: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all' failed
make[1]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

```

Brainiarc7, 2018-06-27:
I'm using the latest version from git, fetched via:

```
git clone https://github.com/intel/libva
```

oliver-sang, 2018-06-27:
how do you build and install libva?

Brainiarc7, 2018-06-27:
I use these steps:

For libva:

```
cd ~/vaapi
git clone https://github.com/intel/libva
cd libva
./autogen.sh --prefix=/usr --libdir=/usr/lib/x86_64-linux-gnu
time make -j$(nproc) VERBOSE=1
sudo make -j$(nproc) install
```

For the libva-utils:

```
cd ~/vaapi
git clone https://github.com/intel/libva-utils 
cd libva-utils
./autogen.sh --prefix=/usr --libdir=/usr/lib/x86_64-linux-gnu
time make -j$(nproc) VERBOSE=1
sudo make -j$(nproc) install

```

oliver-sang, 2018-06-28:
the VASliceParameterBufferHEVCExtension is introduced in https://github.com/intel/libva/commit/d6fd111e2062bb4732db8a05ed55fc01771087b4#diff-dc3ce5eacf55b5dffa4e978ce39dac83
Could you check whether your va/va_dec_hevc.h in your git repo has the declaration?

If so, since you install to /usr, please check whether your /usr/include/va/va_dec_hevc.h is sync with the file in your libva git repo.

If so, please search all va_dec_hevc.h in your system and check whether they are sync.

Brainiarc7, 2018-06-28:
> the VASliceParameterBufferHEVCExtension is introduced in intel/libva@
> d6fd111#diff-dc3ce5eacf55b5dffa4e978ce39dac83

I'll do that and report back.

oliver-sang, 2018-07-25:
@Brainiarc7 any update? I plan to close this issue if you don't have further update. Thanks

Brainiarc7, 2018-07-26:
Hello there,

You may safely close this issue.

Updating libva fixed the issue for me.

closed 2018-07-26 by Brainiarc7

Saentist, 2019-08-09:
I have near same problem with today cloned repo
this is manual with i fallow 
https://gist.github.com/Brainiarc7/4f831867f8e55d35cbcb527e15f9f116
Ubuntu 18.04 x64
4.15.0-55-generic #60-Ubuntu SMP Tue Jul 2 18:22:20 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

```
[ 86%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_encode_base.cpp.o
/home/saen/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp: In member function ‘virtual VAStatus DdiMediaDecode::CreateBuffer(VABufferType, uint32_t, uint32_t, void*, VABufferID*)’:
/home/saen/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:884:14: error: ‘VASubsetsParameterBufferType’ was not declared in this scope
         case VASubsetsParameterBufferType:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/saen/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:884:14: note: suggested alternative: ‘VASliceParameterBufferType’
         case VASubsetsParameterBufferType:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
              VASliceParameterBufferType
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:9974: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_base.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_base.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/saen/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp: In member function ‘uint8_t DdiEncodeBase::VARC2HalRC(uint32_t)’:
/home/saen/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:740:14: error: ‘VA_RC_QVBR’ was not declared in this scope
     else if (VA_RC_QVBR == vaRC)
              ^~~~~~~~~~
/home/saen/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:740:14: note: suggested alternative: ‘VA_RC_VBR’
     else if (VA_RC_QVBR == vaRC)
              ^~~~~~~~~~
              VA_RC_VBR
/home/saen/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:744:14: error: ‘VA_RC_AVBR’ was not declared in this scope
     else if (VA_RC_AVBR == vaRC)
              ^~~~~~~~~~
/home/saen/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:744:14: note: suggested alternative: ‘VA_RC_VBR’
     else if (VA_RC_AVBR == vaRC)
              ^~~~~~~~~~
              VA_RC_VBR
/home/saen/vaapi/workspace/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:752:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1plus: all warnings being treated as errors
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:9998: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_encode_base.cpp.o' failed
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_encode_base.cpp.o] Error 1
CMakeFiles/Makefile2:164: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all' failed
make[1]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

```

any suggestions

Brainiarc7, 2019-08-09:
Your libva installation is out of date.
Did you compile libva from source?

Saentist, 2019-08-09:
> Your libva installation is out of date. Did you compile libva from source?
Yes I fallow your manual

```
cd ~/vaapi
git clone https://github.com/01org/libva
cd libva
./autogen.sh --prefix=/usr --libdir=/usr/lib/x86_64-linux-gnu
time make -j$(nproc) VERBOSE=1
sudo make -j$(nproc) install
sudo ldconfig -vvvv
```

Brainiarc7, 2019-08-09:
I'll try running the rebuild on my test bed and report back to you.

Brainiarc7, 2019-08-09:
In the meanwhile, can you show us the output of vainfo?
Thanks,
Dennis

Saentist, 2019-08-10:
```error: XDG_RUNTIME_DIR not set in the environment.
error: can't connect to X server!
libva info: VA-API version 1.1.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/local/lib/dri/i965_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit```

CPU G3220 if this matter.

Brainiarc7, 2019-08-10:
Your  older libva installation is at `/usr/local/lib`, and that should not
be the case.
What you build (if you followed my guide) should be prefixed to `/usr/`.

Saentist, 2019-08-10:
I fallow manual step-by-step I wont to have working GPU acceleration to transcode live tv to my device.

```
root@B85M-Pro4:/usr/local/lib# ll libva*
-rwxr-xr-x 1 root root    975 дек 23  2018 libva-drm.la*
lrwxrwxrwx 1 root root     20 дек 23  2018 libva-drm.so -> libva-drm.so.2.100.0*
lrwxrwxrwx 1 root root     20 дек 23  2018 libva-drm.so.2 -> libva-drm.so.2.100.0*
-rwxr-xr-x 1 root root  54392 дек 23  2018 libva-drm.so.2.100.0*
-rwxr-xr-x 1 root root   1030 дек 23  2018 libva-glx.la*
lrwxrwxrwx 1 root root     20 дек 23  2018 libva-glx.so -> libva-glx.so.2.100.0*
lrwxrwxrwx 1 root root     20 дек 23  2018 libva-glx.so.2 -> libva-glx.so.2.100.0*
-rwxr-xr-x 1 root root  91344 дек 23  2018 libva-glx.so.2.100.0*
-rwxr-xr-x 1 root root    921 дек 23  2018 libva.la*
lrwxrwxrwx 1 root root     16 дек 23  2018 libva.so -> libva.so.2.100.0*
lrwxrwxrwx 1 root root     16 дек 23  2018 libva.so.2 -> libva.so.2.100.0*
-rwxr-xr-x 1 root root 520408 дек 23  2018 libva.so.2.100.0*
-rwxr-xr-x 1 root root   1016 дек 23  2018 libva-wayland.la*
lrwxrwxrwx 1 root root     24 дек 23  2018 libva-wayland.so -> libva-wayland.so.2.100.0*
lrwxrwxrwx 1 root root     24 дек 23  2018 libva-wayland.so.2 -> libva-wayland.so.2.100.0*
-rwxr-xr-x 1 root root  93416 дек 23  2018 libva-wayland.so.2.100.0*
-rwxr-xr-x 1 root root    997 дек 23  2018 libva-x11.la*
lrwxrwxrwx 1 root root     20 дек 23  2018 libva-x11.so -> libva-x11.so.2.100.0*
lrwxrwxrwx 1 root root     20 дек 23  2018 libva-x11.so.2 -> libva-x11.so.2.100.0*
-rwxr-xr-x 1 root root 180376 дек 23  2018 libva-x11.so.2.100.0*
root@B85M-Pro4:/usr/local/lib# export
declare -x HISTCONTROL="ignoredups:ignorespace"
declare -x HOME="/root"
declare -x LANG="bg_BG.UTF-8"
declare -x LESSCLOSE="/usr/bin/lesspipe %s %s"
declare -x LESSOPEN="| /usr/bin/lesspipe %s"
declare -x LOGNAME="root"
declare -x LS_COLORS="rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:"
declare -x MAIL="/var/mail/root"
declare -x MC_SID="19989"
declare -x MC_TMPDIR="/tmp/mc-root"
declare -x OLDPWD="/usr/local"
declare -x PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin"
declare -x PWD="/usr/local/lib"
declare -x SHELL="/bin/bash"
declare -x SHLVL="2"
declare -x SUDO_COMMAND="/bin/bash"
declare -x SUDO_GID="1000"
declare -x SUDO_UID="1000"
declare -x SUDO_USER="saen"
declare -x TERM="xterm"
declare -x USER="root"
declare -x USERNAME="root"
declare -x XDG_DATA_DIRS="/usr/local/share:/usr/share:/var/lib/snapd/desktop"
```

Brainiarc7, 2019-08-10:
Uninstall that older build first. And follow the documentation on building
libva prefixed to the correct install location.

Saentist, 2019-08-10:
I try to do that but result is same.
Is there any specific steps to do libva uninstalling?