intel/media-driver #215: Fix installing permissions for iHD_drv_video.so

chivakker, 2018-06-27:
With the current build system iHD_drv_video.so is installed with permissions 644, it should be at least 754. Please fix.

oliver-sang, 2018-06-28:
what's the real intention to request this?
If check /usr/lib/x86_64-linux-gnu in ubuntu, most real lib has the mode 644.
If check /usr/lib64 in fedora, yes, most of them has the mode 755.
But what's the difference? Any impact to any user case?
iHD is loaded by libva by designated path/name, I think the read access is enough. Any reason you request to have execute access to it?

chivakker, 2018-06-28:
I don't know what distributions do in terms of modifying permissions, but this is a driver that should have execute permission upon installing.  I don't think distributions are packaging it yet but I'd assume they don't mess up with that.  Other driver install with the right permissions.

chivakker, 2018-12-22:
@XinfengZhang please fix this,  a recent update in gmmlib ABI wasn't caught because this wasn't fixed.

PR #478 (2019-03-07): installing:  install driver with proper permissions
cmake install as PROGRAMS will set the proper permissions
to iHD_drv_video.so file to be 755 in linux

Fixes #215
commit 242bc819 (2019-03-07): installing:  install driver with proper permissions

cmake install as PROGRAMS will set the proper permissions
to iHD_drv_video.so file to be 755 in linux

Fixes #215
closed 2019-03-07 by Sherry-Lin