intel/media-driver #218: [gstreamer-vaapi][hevc_10bit_decode][APL] hevc 10bit decode fail.

feiwan1, 2018-07-03:
System Environment
OS Ubuntu 16.04
Kernel 4.12
libva tag 2.1.0
intel-media-driver 2b6c2a625373df973098bd8105a8744782402e5d
gst-vaapi 1.15.0.1

Reproduce Steps
$ LIBVA_TRACE=./hevc10dec gst-launch-1.0 filesrc location=INITQP_B_Main10_Sony_1.bin ! h265parse ! vaapih265dec ! vaapisink

Error message:
root@yami-apl:/media/hevc_10bit# gst-launch-1.0 filesrc location=INITQP_B_Main10_Sony_1.bin ! h265parse ! vaapih265dec ! vaapisink                             Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'vaapisink0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayX11\)\ vaapidisplayx11-1";
0:00:00.106987875 20557      0x176f6d0 ERROR                  vaapi gstvaapicontext.c:256:config_create: unsupported chroma format (0x00000100)
0:00:00.107071766 20557      0x176f6d0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error -1
0:00:00.107461209 20557      0x176f6d0 ERROR                  vaapi gstvaapicontext.c:256:config_create: unsupported chroma format (0x00000100)
0:00:00.107499522 20557      0x176f6d0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error -1
0:00:00.107733536 20557      0x176f6d0 ERROR                  vaapi gstvaapicontext.c:256:config_create: unsupported chroma format (0x00000100)
0:00:00.107768028 20557      0x176f6d0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error -1
0:00:00.108016894 20557      0x176f6d0 ERROR                  vaapi gstvaapicontext.c:256:config_create: unsupported chroma format (0x00000100)
0:00:00.108050500 20557      0x176f6d0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error -1
0:00:00.108269950 20557      0x176f6d0 ERROR                  vaapi gstvaapicontext.c:256:config_create: unsupported chroma format (0x00000100)
0:00:00.108304201 20557      0x176f6d0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error -1
0:00:00.108572842 20557      0x176f6d0 ERROR                  vaapi gstvaapicontext.c:256:config_create: unsupported chroma format (0x00000100)
0:00:00.108606766 20557      0x176f6d0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error -1
0:00:00.108828437 20557      0x176f6d0 ERROR                  vaapi gstvaapicontext.c:256:config_create: unsupported chroma format (0x00000100)
0:00:00.108862049 20557      0x176f6d0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error -1
0:00:00.109187171 20557      0x176f6d0 ERROR                  vaapi gstvaapicontext.c:256:config_create: unsupported chroma format (0x00000100)
0:00:00.109223104 20557      0x176f6d0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error -1
0:00:00.109643549 20557      0x176f6d0 ERROR                  vaapi gstvaapicontext.c:256:config_create: unsupported chroma format (0x00000100)
0:00:00.109685115 20557      0x176f6d0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error -1
0:00:00.109938652 20557      0x176f6d0 ERROR                  vaapi gstvaapicontext.c:256:config_create: unsupported chroma format (0x00000100)
0:00:00.109971842 20557      0x176f6d0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error -1
0:00:00.110227903 20557      0x176f6d0 ERROR                  vaapi gstvaapicontext.c:256:config_create: unsupported chroma format (0x00000100)
0:00:00.110269254 20557      0x176f6d0 ERROR            vaapidecode gstvaapidecode.c:755:gst_vaapidecode_handle_frame: decode error -1
ERROR: from element /GstPipeline:pipeline0/GstVaapiDecode_h265:vaapidecode_h265-0: Decoding error
Additional debug info:
gstvaapidecode.c(764): gst_vaapidecode_handle_frame (): /GstPipeline:pipeline0/GstVaapiDecode_h265:vaapidecode_h265-0:
Decode error -1
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...

feiwan1, 2018-07-03:
[hevc10dec.164437.thd-0x00005072.log](https://github.com/intel/media-driver/files/2158430/hevc10dec.164437.thd-0x00005072.log)

By using any hevc main10 stream can reproduce this issue.

XinfengZhang, 2018-07-06:
please help to check whether #221  can fix the issue.

XinfengZhang, 2018-07-19:
@feiwan1 could you help to check #221?

feiwan1, 2018-08-22:
Hi @XinfengZhang , sorry for late response. I tried your patch on KBL, but got same error log.

wangzj0601, 2019-03-14:
this is a regression issue.
regression commit:
d7e8c7c sw swizzle surface in vaGetImage

this issue can not be duplicated with gst-msdk on APL.

feiwan1, 2019-08-29:
Tested with commit 43df6f55da70d2746e43ed1262b3d2e25c6d0979. Looks good. Close this.

closed 2019-08-29 by feiwan1