intel/media-driver #230: [gstreamer-vaapi][vpp] scaling vpp fail

wangzj0601, 2018-07-13:
Test Env:
============================================
Platform:               APL/SKL
Arch:          		x86_64
Kernel:			4.16.0
libva			commit b3be72a5a110880f70626d7c3bed953cdde124b2
media_driver		commit 4a85bb18ecc9a18b7cf592d405c8c2fa1c7aad06
gstreamer               master branch 9722e896446ed052623d795d8511140fbca7f023
gst-vaapi 		master branch f5eb4faa5914f3745820e557ac2401a7d738be66

Reproduce Steps:
============================================
1. build enc as above lists
2. gst-launch-1.0 filesrc location=/media//encoderbitstreams/gstvpp//768x480.I420 '!' videoparse format=i420 width=768 height=480 framerate=1 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapipostproc width=1920 height=1088 '!' videoconvert '!' video/x-raw,format=I420 '!' checksumsink2 frame-checksum=false dump-output=true dump-location=scale.I420
3. error message
0:00:00.140513466  2716      0x12729e0 ERROR          vaapipostproc gstvaapipostproc.c:872:gst_vaapipostproc_process_vpp:<vaapipostproc0> failed to apply VPP filters (error 2)
ERROR: from element /GstPipeline:pipeline0/GstVideoParse:videoparse0/GstRawVideoParse:inner_rawvideoparse: Internal data stream error.
Additional debug info:
gstbaseparse.c(3656): gst_base_parse_loop (): /GstPipeline:pipeline0/GstVideoParse:videoparse0/GstRawVideoParse:inner_rawvideoparse:
streaming stopped, reason error (-5)

Jister, 2018-09-04:
Hi, what addition package need to be installed to setup the environment? When I reproduce the issue, it shows no element "checksumsink2".

wangzj0601, 2018-09-04:
you should install "gst-checksumsink", the repo is "https://github.com/ceyusa/gst-checksumsink"
or you can just run cmd without "checksumsink".
"gst-launch-1.0 filesrc location=/media//encoderbitstreams/gstvpp//768x480.I420 '!' videoparse format=i420 width=768 height=480 framerate=1 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapipostproc width=1920 height=1088 '!' videoconvert '!' video/x-raw,format=I420 '!' filesink location=scale.I420"

wangzj0601, 2018-09-04:
sorry for my mistake.

Jister, 2018-09-14:
@wangzj0601 why do you need to convert from I420 to NV12, then to I420, rather than do scaling directly? I saw the error log, it shows "streaming stopped, reason not-negotiated (-4)", it seems the elements of gstreamer are not negotiated while using your command. I tried to remove the format conversion from your command, the scaling can work well.

FurongZhang, 2018-10-17:
Pending on submitter to verify that.

uartie, 2018-10-17:
@FurongZhang without format conversion, I see this issue.  Perhaps try these commands:

This command fails (i.e. format=I420):
```
LIBVA_DRIVER_NAME=iHD GST_VAAPI_ALL_DRIVERS=1 gst-launch-1.0 -vf \
  videotestsrc num-buffers=10 ! video/x-raw,format=I420,width=768,height=480 \
  ! vaapipostproc width=1920 height=1088 ! filesink location=output.I420
```

This command passes (i.e. format=NV12):
```
LIBVA_DRIVER_NAME=iHD GST_VAAPI_ALL_DRIVERS=1 gst-launch-1.0 -vf \
  videotestsrc num-buffers=10 ! video/x-raw,format=NV12,width=768,height=480 \
  ! vaapipostproc width=1920 height=1088 ! filesink location=output.I420
```

Oddly, this command passes for me (which should be equivalent to initial reported command):
```
LIBVA_DRIVER_NAME=iHD GST_VAAPI_ALL_DRIVERS=1 gst-launch-1.0 -vf \
  videotestsrc num-buffers=10 ! video/x-raw,format=I420,width=768,height=480 \
  ! videoconvert ! video/x-raw,format=NV12 ! vaapipostproc width=1920 height=1088 \
  ! videoconvert ! video/x-raw,format=I420 ! filesink location=output.I420
```

All commands work on i965 driver.

NOTE: This is my current stack
```
libva (master) heads/master-0-gf0dfe5e762e7 
gmmlib (master) heads/master-0-g456b7c0354a4 
intel-media-driver (master) heads/master-0-g5adf28710aa5 
gstreamer (master) heads/master-0-g8f761cd7114b 
gstreamer-vaapi (master) heads/master-0-g82872f423445
```

Jister, 2018-11-07:
Hi, @uartie , for the first failed test case:
```
LIBVA_DRIVER_NAME=iHD GST_VAAPI_ALL_DRIVERS=1 gst-launch-1.0 -vf \
  videotestsrc num-buffers=10 ! video/x-raw,format=I420,width=768,height=480 \
  ! vaapipostproc width=1920 height=1088 ! filesink location=output.I420
```
the failed reason is that when create I420 surface, gstreamer-vaapi didn't set fourcc of I420 in attribute list, and it created a NV12 surface.

ceyusa, 2018-11-07:
> the failed reason is that when create I420 surface, gstreamer-vaapi didn't set fourcc of I420 in attribute list, and it created a NV12 surface.

It would be interesting, in vaapipostproc, to set the upstream color format as the first option when negotiating downstream, instead of the default template, which has NV12 as the first option.

wangzj0601, 2018-12-07:
test with the latest gstreamer and media driver, this issue can not be duplicated with below env, so close it.

libva	commit 285267586a3d4db0e721d30d4a5f5f9fe6a3c913
media_driver	commit 5976380
gstreamer master branch c45b386b54aea4e707a8be5cbf79dc43623c4614
gst-vaapi master branch c728fb4ff6c3b4fc9eb9b43fcea4a22bc15fd678

closed 2018-12-07 by wangzj0601