intel/media-driver #231: [gstreamer-vaapi][vpp] two cases fail during doing csc

wangzj0601, 2018-07-13:
Test Env:
============================================
Platform:               APL/SKL
Arch:          		x86_64
Kernel:			4.16.0
libva			commit b3be72a5a110880f70626d7c3bed953cdde124b2
media_driver		commit 4a85bb18ecc9a18b7cf592d405c8c2fa1c7aad06
gstreamer               master branch 9722e896446ed052623d795d8511140fbca7f023
gst-vaapi 		master branch f5eb4faa5914f3745820e557ac2401a7d738be66

Reproduce Steps:
============================================
1. build enc as above lists
2. gst-launch-1.0 filesrc location=/media//encoderbitstreams/gstvpp//768x480.I420 '!' videoparse format=i420 width=768 height=480 framerate=1 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapipostproc format=7 '!' videoconvert '!' video/x-raw,format=I420 '!' checksumsink2 frame-checksum=false dump-output=true dump-location=cscrgbx.I420
3. error message
0:00:00.127384980  9391       0xfa59e0 ERROR       vaapivideomemory gstvaapivideomemory.c:736:gst_video_info_update_from_surface: Cannot create a VA derived image from surface 0x7f8f84002c00

or 
gst-launch-1.0 filesrc location=/media//encoderbitstreams/gstvpp//768x480.I420 '!' videoparse format=i420 width=768 height=480 framerate=1 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapipostproc format=3 '!' videoconvert '!' video/x-raw,format=I420 '!' checksumsink2 frame-checksum=false dump-output=true dump-location=cscyv12.I420
this case will core dumped probably(failure rate: 2/10

feiwan1, 2018-07-13:
Hi @wangzj0601, this issue looks same with https://github.com/intel/media-driver/issues/155. we can use #155 only to track the vpp issue if possible.

wangzj0601, 2018-07-13:
duplicate with #155, so close it.

closed 2018-07-13 by wangzj0601