intel/media-driver #232: [gstreamer-vaapi][jpeg-enc] the parameter "quality" does not work when set the different value

wangzj0601, 2018-07-13:
Test Env:
Platform: APL/SKL
Arch: x86_64
Kernel:	4.16.0
libva	commit b3be72a5a110880f70626d7c3bed953cdde124b2
media_driver	commit 4a85bb1
gstreamer master branch 9722e896446ed052623d795d8511140fbca7f023
gst-vaapi master branch f5eb4faa5914f3745820e557ac2401a7d738be66

Reproduce Steps:
1. build enc as above lists
2. gst-launch-1.0 -f filesrc location=/media//encoderbitstreams/jpeg//720x480.jpg_720x480.I420 '!' videoparse format=i420 width=720 height=480 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapijpegenc quality=40 '!' jpegparse '!' filesink location=./720x480.jpg_720x480.I420_40.jpeg
3. change the quality value,  find that the md5 of encoded file is always same.

XinfengZhang, 2018-07-19:
jpeg quality is determined by Qmatrix. if there are no Qmatrix,, driver will calculate qmatrix with the "quality" parameter, so , if you want "quality" take effect, two ways, 1 is calculate qmatrix in middle ware, 2 is dont send qmatrix. I will check the issue later.

wangzj0601, 2018-12-04:
this issue has been fixed with the latest driver, so close this issue

closed 2018-12-04 by wangzj0601