intel/media-driver #235: iHD_drv_video.so has no function __vaDriverInit_0_32

nehudesi, 2018-07-16:
I am trying to run Hardware Acceleration using iHD driver on Intel Apollo lake.
I have setup Ubuntu 16.04 machine and followed https://github.com/intel/media-driver steps to setup media-driver.

My machine Info:
model name    : Intel(R) Celeron(R) CPU N3350 @ 1.10GHz
00:02.0 VGA compatible controller: Intel Corporation Device 5a85 (rev 0b) (prog-if 00 [VGA controller])

However, vainfo gives me below error:

$ vainfo
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva error: /usr/local/lib/dri/iHD_drv_video.so has no function __vaDriverInit_0_32
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit

Any idea, what I am missing?

Any help/suggestion would be appreciated.

dvrogozh, 2018-07-16:
> libva info: VA-API version 0.39.0

That's what is wrong: you are trying to use previous version of libva (API 0.x series) while this driver is compatible with API 1.x. Since this is a major change, there is no backward compatibility. Make sure that you rebuilt your application against newer version of libva and use it at runtime. If you are on Ubuntu you can substitute system version of libva with your own build if you will configure as:
```
./autogen.sh --prefix=/usr --libdir=/usr/lib/x86_64-linux-gnu
```

nehudesi, 2018-07-16:
I am new to media-driver, hence sorry for noise if this was insanely simple.
Thank You dvrogozh , I am able to pass that error and now my vainfo shows loading of iHD driver.

Now that I have setup ready, do you have any recommendation on how to test/play video through Hardware acceleration. I mean any simple app to start with?

Thanks...

XinfengZhang, 2018-07-19:
you can use ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 xxxxxxxxx,  or libva-utils https://github.com/intel/libva-utils , there are sample codes in encode/decode folder

nehudesi, 2018-07-19:
Thank you XinfengZhang!

djmv, 2018-07-24:
@dvrogozh what is exactly the application to rebuild ? . I have the same issue.  I built libva and intel-vaapi. 

$ vainfo
libva info: VA-API version 1.1.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva error: /usr/local/lib/dri/iHD_drv_video.so has no function __vaDriverInit_1_0
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit

Please, help me !

dvrogozh, 2018-07-24:
@djmv. No, your issue is different. This time you have vaapi (libva) of the later version (1.1) than the driver (1.0). You need either downgrade libva to the same version as a driver or update the driver to match vaapi version.

closed 2019-08-28 by XinfengZhang

liswin, 2020-01-30:
@dvrogozh 
```
libva info: VA-API version 0.40.0                                               
libva info: va_getDriverName() returns 0                                        
libva info: User requested driver 'iHD'                                         
libva info: Trying to open /opt/intel/mediasdk/lib64/iHD_drv_video.so           
libva error: /opt/intel/mediasdk/lib64/iHD_drv_video.so has no function __vaDriv
erInit_0_32                                                                     
libva info: va_openDriver() returns -1                                          
vaInitialize failed with error code -1 (unknown libva error),exit    
```
What is the application that I've to rebuild? Could you please guide me?