intel/media-driver #237: Driver will truncate some important value to a smaller range

xuguangxin, 2018-07-17:
1. apply this https://github.com/intel/media-driver/pull/236
2. build driver with this command
<pre> CFLAGS="-Werror=bitfield-constant-conversion -w" CXXFLAGS="-Werror=bitfield-constant-conversion -w" cmake ../media-driver -DCMAKE_INSTALL_PREFIX=/usr -DMEDIA_VERSION="2.0.0" -DBS_DIR_GMMLIB=`pwd`/../gmmlib/Source/GmmLib/ -DBS_DIR_COMMON=`pwd`/../gmmlib/Source/Common/ -DBS_DIR_INC=`pwd`/../gmmlib/Source/inc/ -DBS_DIR_MEDIA=`pwd`/../media-driver -DCMAKE_C_COMPILER=clang -DCMAKE_CXX_COMPILER=clang++ </pre>
you will see error like this.

```
/sdc/github/media-driver/media_driver/agnostic/gen9/hw/mhw_vebox_hwcmd_g9_X.cpp:165:54: error: implicit truncation from 'mhw_vebox_g9_X::VEBOX_CCM_STATE_CMD::C5' to bit-field changes value from 2097100 to 131020 [-Werror,-Wbitfield-constant-conversion]
/sdc/github/media-driver/media_driver/agnostic/gen9/hw/mhw_vebox_hwcmd_g9_X.cpp:174:54: error: implicit truncation from 'mhw_vebox_g9_X::VEBOX_CCM_STATE_CMD::C6' to bit-field changes value from 2097140 to 131060 [-Werror,-Wbitfield-constant-conversion]
/sdc/github/media-driver/media_driver/agnostic/gen10/codec/hal/codechal_encode_hevc_g10.cpp:6514:48: error: implicit truncation from 'int' to bit-field changes value from -10 to 65526 [-Werror,-Wbitfield-constant-conversion]
/sdc/github/media-driver/media_driver/agnostic/gen10/codec/hal/codechal_vdenc_hevc_g10.cpp:1895:72: error: implicit truncation from 'uint16_t' (aka 'unsigned short') to bit-field changes value from 65535 to 32767 [-Werror,-Wbitfield-constant-conversion]

```
all four are important, the third one is mostimportant , the driver will convert a -10 to an int max...

liangali, 2018-08-27:
@yangnin, could you help on this?

yangnin, 2018-09-03:
Fix with PR#300, Please help to review.
https://github.com/intel/media-driver/pull/300

closed 2018-11-15 by yangnin