intel/media-driver #238: Build pending at linux/ult/ult_app/ddi_test_decode.cpp

jwang11, 2018-07-18:
commit 358d05e0c9ac957c696ab84562b5e8cbaa40d684
model name      : Intel(R) Core(TM) i5-7200U CPU @ 2.50GHz
ClearLinux  version: 23730

Steps:
Following README to build, however build pending at final stage when doing unit testing, details see below. I have to CTL-C to interrupt.

Running devult...
[==========] Running 43 tests from 13 test cases.
[----------] Global test environment set-up.
[----------] 1 test from MediaCapsDdiTest
[ RUN      ] MediaCapsDdiTest.DecodeEncodeProfile
DRM_IOCTL_I915_GEM_APERTURE failed: Inappropriate ioctl for device
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [25]
param: 4, val: 0
/home/jwang/devel/media_driver/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:99: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 1
Platform = SKL, Failed function = m_driverLoader.InitDriver

/home/jwang/devel/media_driver/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:103: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: -1
Platform = SKL, Failed function = Test_QueryConfigProfiles

/home/jwang/devel/media_driver/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:109: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaRenderPicture

/home/jwang/devel/media_driver/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:109: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaRenderPicture

/home/jwang/devel/media_driver/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:109: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaRenderPicture

/home/jwang/devel/media_driver/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:114: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaEndPicture

^Cmake[2]: *** [media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/build.make:63: RunULT] Interrupt
make: *** [Makefile:152: all] Interrupt

XinfengZhang, 2018-07-19:
thanks for reporting , will check it ASAP

walter-bai, 2018-07-23:
> DRM_IOCTL_I915_GEM_APERTURE failed: Inappropriate ioctl for device

This issue is caused because function mosdrmIoctl failed.

There are 2 versions of mosdrmIoctl, 1 for driver, which is in iHD_drv_video.so and will perform a system call. Another is in libdrm_mock.so, which is a mock version and will only be used when running ULT. The mock version should not fail.

Normally a path of libdrm_mock.so will be assigned to LD_PRELOAD before running ULT, see media_driver/linux/ult/ult_app/CMakeLists.txt, so ULT will use mock version of mosdrmIoctl.
```
add_custom_command(
            TARGET RunULT
            POST_BUILD
            COMMAND LD_PRELOAD=../libdrm_mock/libdrm_mock.so ./devult ../../../${LIB_NAME}.so
            WORKING_DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}
            COMMENT "Running devult...")
```
Could you first confirm that which version of mosdrmIoctl is called? You can use printf to add some messages in mosdrmIoctl of media_driver/linux/ult/libdrm_mock/xf86drm_mock.c.

jwang11, 2018-07-25:
In build environment, I manually execute below command, still get same error
```
LD_PRELOAD=../libdrm_mock/libdrm_mock.so ./devult ../../../iHD_drv_video.so
```

walter-bai, 2018-07-26:
I use your CFLAGS on Ubuntu 16.
```
export LD_DEBUG=all
make -j8 2>&1 | grep mosdrmIoctl
```
Get the following message:
> binding file ../../../iHD_drv_video.so [0] to ../libdrm_mock/libdrm_mock.so [0]: normal symbol `mosdrmIoctl'

It's expected, so CFLAGS is not the root cause.
I see the following in man page of LD_PRELOAD, could you check whether issue exists when building driver as root?

> For set-user-ID/set-group-ID ELF binaries, preload pathnames containing slashes are ignored, and libraries in the standard search directories are loaded only if the set-user-ID permission bit is enabled on the library file.

jwang11, 2018-07-26:
@walter-bai root user doesn't help. Sorry, I gave a wrong hint about CFLAGS previously. For iHD build, it is CXXFLAGS take effect, not CFLAGS. In ClearLinux,
```
CXXFLAGS=-g -O3 -feliminate-unused-debug-types  -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=32 -Wformat -Wformat-security -Wl,--copy-dt-needed-entries -m64  -fasynchronous-unwind-tables -Wp,-D_REENTRANT -ftree-loop-distribute-patterns -Wl,-z -Wl,now -Wl,-z -Wl,relro -fno-semantic-interposition -ffat-lto-objects  -fno-signed-zeros -fno-trapping-math  -fassociative-math -Wl,-sort-common -Wl,--enable-new-dtags -fvisibility-inlines-hidden -Wl,--enable-new-dtags
```
After I update it to 
```
CXXFLAGS=-g -O3
```
Everything is OK. Therefore, CXXFLAGS is really relevant  to the issue. However I haven't looked into it to identify exactly the culprit.

walter-bai, 2018-07-27:
On my Ubuntu 16, the culprit is 

> -fno-semantic-interposition

Here is some introduction of -fno-semantic-interposition from GCC manual:

> With -fno-semantic-interposition the compiler assumes that if interposition happens for functions the overwriting function will have precisely the same semantics (and side effects).

This restricts interposing of mock version of mosdrmIoctl by demanding same semantics and side effects with the non-mock one and may be the reason why mosdrmIoctl is not interposed.

jwang11, 2018-07-31:
For security reason, clear linux can't drop -fno-semantic-interposition to compromise. So, a workaround way to build media driver in ClearLinux is to append -DBYPASS_MEDIA_ULT=yes as parameter of cmake, sth like
```
cmake ../media-driver \
-DCMAKE_INSTALL_PREFIX=/usr \
-DBYPASS_MEDIA_ULT=yes \
-DMEDIA_VERSION="2.0.0" \
-DBS_DIR_GMMLIB=`pwd`/../gmmlib/Source/GmmLib/ \
-DBS_DIR_COMMON=`pwd`/../gmmlib/Source/Common/ \
-DBS_DIR_INC=`pwd`/../gmmlib/Source/inc/ \
-DBS_DIR_MEDIA=`pwd`/../media-driver $@
```

XinfengZhang, 2020-07-03:
build option -fno-semantic-interposition block mock function, and -DBYPASS_MEDIA_ULT=yes could be used to help bypass ULT test, it could be closed

closed 2020-07-03 by XinfengZhang