intel/media-driver #243: The gmmlib install directory for this package breaks computre-runtime builds

Brainiarc7, 2018-07-21:
See this issue: https://github.com/intel/compute-runtime/issues/61

oliver-sang, 2018-07-25:
media will follow neo's method to convert to use gmmlib dynamic lib.

yangnin, 2018-09-03:
@Brainiarc7 Do u have tied the lastest  media driver? It has been switched to dynamic gmmlib.

Brainiarc7, 2018-09-07:
Yes, the issue is fixed in the latest driver.

I can now close this issue.

closed 2018-09-07 by Brainiarc7

dseeker, 2018-10-20:
@Brainiarc7 I'm stuck at "compute runtime project"
time make -j$(nproc) package VERBOSE=1

what is the procedure to update the driver and get rid of the 

> /hw_info.h:9:10: fatal error: gtsysinfo.h: No such file or directory

 errors?

Brainiarc7, 2018-10-20:
That's because the Intel Graphics Compiler (IGC) is not installed.

Show us your build steps.

dseeker, 2018-10-20:
was following https://gist.github.com/Brainiarc7/4f831867f8e55d35cbcb527e15f9f116 line by line:

> Next, build and install the compute runtime project. Start by creating a separate build directory for it:
> mkdir -p ~/intel-compute-runtime/workspace/build_icr
> cd ~/intel-compute-runtime/workspace/build_icr
> cmake -DCMAKE_BUILD_TYPE=Release ../neo
> time make -j$(nproc) package VERBOSE=1

Brainiarc7, 2018-10-20:
With that mentioned, were you able to build ICR?

It's a prerequisite for building IGC.

dseeker, 2018-10-20:
```
**~/intel-compute-runtime/workspace/build_icr # cmake -DCMAKE_BUILD_TYPE=Release ../neo**
-- Release build configuration
-- i915 includes dir: /root/intel-compute-runtime/workspace/neo/third_party/uapi
-- Khronos OpenCL headers dir: /root/intel-compute-runtime/workspace/neo/third_party/opencl_headers
-- Khronos OpenGL headers dir: /root/intel-compute-runtime/workspace/neo/third_party/opengl_headers
-- Third party dir: /root/intel-compute-runtime/workspace/third_party
-- IGC include dirs: /usr/include/igc;/usr/include/igc/cif;/usr/include/igc/ocl_igc_shared/executable_format;/usr/include/igc/ocl_igc_shared/device_enqueue
-- GmmLib include dirs: /usr/include/igdgmm;/usr/include/igdgmm/GmmLib;/usr/include/igdgmm/GmmLib/inc;/usr/include/igdgmm/inc;/usr/include/igdgmm/inc/common;/usr/include/igdgmm/util
-- Computed version major is: 18
-- Computed version minor is: 42
-- All supported platforms:  CNL SKL KBL GLK CFL BXT BDW
-- All tested platforms:  CNL SKL KBL GLK CFL BXT BDW
-- Default supported platform: SKL
-- Default tested platform: SKL
-- Default tested family name: SKLFamily
-- Found ccache: /usr/bin/ccache
-- Using libva
-- GTest repeat count set to 1
-- GTest shuffle set to --gtest_shuffle;--gtest_random_seed=0
-- GTest exception options set to --gtest_catch_exceptions=1
-- GTest filter for mt tests:-AsyncGPUoperations.MapBufferAfterWriteBuffer
-- Configuring done
-- Generating done
-- Build files have been written to: /root/intel-compute-runtime/workspace/build_icr
```

```
 **~/intel-compute-runtime/workspace/build_icr # time make -j$(nproc) package VERBOSE=1**
In file included from /root/intel-compute-runtime/workspace/neo/runtime/gen8/hw_info_gen8.h:9:0,
                 from /root/intel-compute-runtime/workspace/neo/unit_tests/gen8/cmd_parse_gen8.cpp:8:
/root/intel-compute-runtime/workspace/neo/runtime/helpers/hw_info.h:9:10: fatal error: gtsysinfo.h: No such file or directory
 #include "gtsysinfo.h"
          ^~~~~~~~~~~~~
compilation terminated.
unit_tests/libult/CMakeFiles/igdrcl_libult.dir/build.make:254: recipe for target 'unit_tests/libult/CMakeFiles/igdrcl_libult.dir/__/gen8/cmd_parse_gen8.cpp.o' failed
make[2]: *** [unit_tests/libult/CMakeFiles/igdrcl_libult.dir/__/gen8/cmd_parse_gen8.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[  4%] Building CXX object unit_tests/mocks/CMakeFiles/igdrcl_mocks.dir/mock_csr.cpp.o
cd /root/intel-compute-runtime/workspace/build_icr/unit_tests/mocks && ccache /usr/bin/c++  -DCIF_HEADERS_ONLY_BUILD -DCL_TARGET_OPENCL_VERSION=210 -DCL_USE_DEPRECATED_OPENCL_1_1_APIS -DCL_USE_DEPRECATED_OPENCL_1_2_APIS -DCL_USE_DEPRECATED_OPENCL_2_0_APIS -DDEFAULT_PLATFORM=SKL -DDEFAULT_TEST_FAMILY_NAME=SKLFamily -DDEFAULT_TEST_PLATFORM=SKL -DGL_TARGET_OPENGL_VERSION=210 -DGMM_LIB_DLL -DGMM_OCL -DLIBVA -DMOCKABLE_VIRTUAL=virtual -DSUPPORT_BDW -DSUPPORT_BXT -DSUPPORT_CFL -DSUPPORT_CNL -DSUPPORT_GEN10 -DSUPPORT_GEN8 -DSUPPORT_GEN9 -DSUPPORT_GLK -DSUPPORT_KBL -DSUPPORT_SKL -DTESTS_BDW -DTESTS_BXT -DTESTS_CFL -DTESTS_CNL -DTESTS_GEN10 -DTESTS_GEN8 -DTESTS_GEN9 -DTESTS_GLK -DTESTS_KBL -DTESTS_SKL -I/root/intel-compute-runtime/workspace/neo -I/root/intel-compute-runtime/workspace/build_icr -I/root/intel-compute-runtime/workspace/neo/runtime/sku_info/definitions -I/root/intel-compute-runtime/workspace/neo/runtime/os_interface/definitions -I/root/intel-compute-runtime/workspace/neo/runtime/command_stream/definitions -I/root/intel-compute-runtime/workspace/neo/runtime/gen_common/reg_configs -I/root/intel-compute-runtime/workspace/neo/runtime/gmm_helper -I/root/intel-compute-runtime/workspace/neo/runtime/gmm_helper/client_context -I/root/intel-compute-runtime/workspace/neo/runtime/gmm_helper/gmm_memory -I/root/intel-compute-runtime/workspace/neo/runtime/instrumentation -I/root/intel-compute-runtime/workspace/neo/third_party/gtest -I/root/intel-compute-runtime/workspace/neo/third_party/opencl_headers -I/root/intel-compute-runtime/workspace/neo/third_party/opengl_headers -I/root/intel-compute-runtime/workspace/neo/runtime/gen_common -I/usr/include/igc -I/usr/include/igc/cif -I/usr/include/igc/ocl_igc_shared/executable_format -I/usr/include/igc/ocl_igc_shared/device_enqueue -I/root/intel-compute-runtime/workspace/third_party -I/usr/include/igdgmm -I/usr/include/igdgmm/GmmLib -I/usr/include/igdgmm/GmmLib/inc -I/usr/include/igdgmm/inc -I/usr/include/igdgmm/inc/common -I/usr/include/igdgmm/util -I/root/intel-compute-runtime/workspace/neo/runtime/os_interface/linux -I/root/intel-compute-runtime/workspace/neo/third_party/uapi -I/root/intel-compute-runtime/workspace/neo/elf  -ftemplate-depth=1024 -Wall -Wempty-body -Wignored-qualifiers -Wtype-limits -Wuninitialized -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wno-unused-local-typedefs -Wno-unused-but-set-variable -Wclobbered -Wimplicit-fallthrough=4 -Wno-noexcept-type -Werror -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -O3 -DNDEBUG -fPIC   -std=gnu++14 -o CMakeFiles/igdrcl_mocks.dir/mock_csr.cpp.o -c /root/intel-compute-runtime/workspace/neo/unit_tests/mocks/mock_csr.cpp
In file included from 
/root/intel-compute-runtime/workspace/neo/runtime/helpers/hw_info.h:9:10: fatal error: gtsysinfo.h: No such file or directory
 #include "gtsysinfo.h"
          ^~~~~~~~~~~~~
compilation terminated.
[... cut ...]
make[2]: *** [unit_tests/mocks/CMakeFiles/igdrcl_mocks.dir/mock_context.cpp.o] Error 1
make[2]: Leaving directory '/root/intel-compute-runtime/workspace/build_icr'
CMakeFiles/Makefile2:6612: recipe for target 'unit_tests/mocks/CMakeFiles/igdrcl_mocks.dir/all' failed
make[1]: *** [unit_tests/mocks/CMakeFiles/igdrcl_mocks.dir/all] Error 2
make[1]: Leaving directory '/root/intel-compute-runtime/workspace/build_icr'
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

real    0m0.336s
user    0m2.019s
sys     0m0.515s

```

dseeker, 2018-10-20:
@Brainiarc7 apparently so, there are some warnings but it completes
```
 cmake -DIGC_OPTION__OUTPUT_DIR=../igc-install/Release \
>     -DCMAKE_BUILD_TYPE=Release -DIGC_OPTION__ARCHITECTURE_TARGET=Linux64 \
>     ../igc/IGC
CMake Warning (dev) at CMakeLists.txt:2058 (set):
  Cannot set "IGC_LIBRARY_NAME": current scope has no parent.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:2067 (set):
  Cannot set "FCL_LIBRARY_NAME": current scope has no parent.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- ================================ IGC Project ================================
-- Build type:  Release (single-configuration generator)
-- Build tools: OFF
-- LLVM dir:
--
-- Output directory:
--    "/root/intel-compute-runtime/workspace/igc/igc-install/Release"
-- Architecture:
--  - target: Linux64    (detected: Linux64)
--  - host:   Linux64    (detected: Linux64)
--
-- Cross-compilation needed: NO
-- Cross-compilation set:    FALSE
--
-- Advanced:
--  - Link BiF resources:              ON
--  - Building Windows Universal:
-- =============================================================================
CMake Warning at CMakeLists.txt:2206 (message):
  not defined, defaulting to y: UFO_VK

-- [IGC] LLVM targets are not defined. Try to use LLVM from standard known locations
-- [IGC] LLVM compilation from /root/intel-compute-runtime/workspace/igc/IGC/../../llvm_patches
-- LLVM_SOURCE_URL   = /root/intel-compute-runtime/workspace/llvm_patches/../llvm_source
-- LLVM_OPTIONS      = -DLLVM_TARGETS_TO_BUILD=;-DLLVM_BUILD_TOOLS=true;-DLLVM_INCLUDE_TOOLS=true;-DLLVM_INSTALL_UTILS=false;-DLLVM_INCLUDE_UTILS=false;-DLLVM_BUILD_EXAMPLES=false;-DLLVM_INCLUDE_EXAMPLES=false;-DLLVM_BUILD_TESTS=false;-DLLVM_INCLUDE_TESTS=false;-DLLVM_APPEND_VC_REV=false;-DLLVM_ENABLE_THREADS=true;-DLLVM_ENABLE_CXX1Y=false;-DLLVM_ENABLE_PIC=true;-DLLVM_ENABLE_WARNINGS=true;-DLLVM_ENABLE_PEDANTIC=true;-DLLVM_ENABLE_WERROR=false;-DLLVM_ABI_BREAKING_CHECKS=FORCE_OFF;-DLLVM_BUILD_RUNTIME=true;-DLLVM_ENABLE_TERMINFO=false;-DLLVM_ENABLE_EH=true;-DLLVM_ENABLE_RTTI=true;-DLLVM_BUILD_32_BITS=false
-- LLVM_BUILD_TYPE   = Release
-- [LLVM] PYTHON_EXECUTABLE = /usr/bin/python2
-- [LLVM] Clearing build system compilation flags
-- Native target X86 is not selected; lli will not JIT code
-- Threads enabled.
-- Doxygen disabled.
-- Sphinx disabled.
-- Go bindings disabled.
-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH)
-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH)
-- OCaml bindings disabled.
-- LLVM host triple: x86_64-unknown-linux-gnu
-- LLVM default target triple: x86_64-unknown-linux-gnu
-- Building with -fPIC
-- Constructing LLVMBuild project information
-- [LLVM] Restoring build system compilation flags
-- SPIRV-Tools: nosetests found - python support code will be tested
-- Precompiled common clang bundle /root/intel-compute-runtime/workspace/igc/IGC/../Clang/Prebuilt/linux-ubuntu/Release/64/clang.7z does not exist. Try to compile from sources.
CMake Warning (dev) in BiFModule/CMakeLists.txt:
  A logical block opening on the line

    /root/intel-compute-runtime/workspace/igc/IGC/BiFModule/CMakeLists.txt:166 (if)

  closes on the line

    /root/intel-compute-runtime/workspace/igc/IGC/BiFModule/CMakeLists.txt:169 (endif)

  with mis-matching arguments.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Trigger common clang compilation from /root/intel-compute-runtime/workspace/common_clang to /root/intel-compute-runtime/workspace/igc/igc-install/Release/clang/linux-ubuntu64
-- LLVM_SOURCE_URL        = /root/intel-compute-runtime/workspace/common_clang/../llvm_source
[... 141 lines cut ...]
make[2]: Leaving directory '/root/intel-compute-runtime/workspace/build_igc'
[ 94%] Built target Compiler
make -f CMakeFiles/igc_dll.dir/build.make CMakeFiles/igc_dll.dir/depend
make[2]: Leaving directory '/root/intel-compute-runtime/workspace/build_igc'
make -f CMakeFiles/fcl_dll.dir/build.make CMakeFiles/fcl_dll.dir/build
make[2]: Entering directory '/root/intel-compute-runtime/workspace/build_igc'
make[2]: Nothing to be done for 'CMakeFiles/fcl_dll.dir/build'.
make[2]: Leaving directory '/root/intel-compute-runtime/workspace/build_igc'
[ 96%] Built target fcl_dll
make[2]: Entering directory '/root/intel-compute-runtime/workspace/build_igc'
cd /root/intel-compute-runtime/workspace/build_igc && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /root/intel-compute-runtime/workspace/igc/IGC /root/intel-compute-runtime/workspace/igc/IGC /root/intel-compute-runtime/workspace/build_igc /root/intel-compute-runtime/workspace/build_igc /root/intel-compute-runtime/workspace/build_igc/CMakeFiles/igc_dll.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/root/intel-compute-runtime/workspace/build_igc'
make -f CMakeFiles/igc_dll.dir/build.make CMakeFiles/igc_dll.dir/build
make[2]: Entering directory '/root/intel-compute-runtime/workspace/build_igc'
make[2]: Nothing to be done for 'CMakeFiles/igc_dll.dir/build'.
make[2]: Leaving directory '/root/intel-compute-runtime/workspace/build_igc'
[100%] Built target igc_dll
make[1]: Leaving directory '/root/intel-compute-runtime/workspace/build_igc'
/usr/bin/cmake -E cmake_progress_start /root/intel-compute-runtime/workspace/build_igc/CMakeFiles 0

real    0m1.305s
user    0m1.767s
sys     0m0.534s

```

Brainiarc7, 2018-10-20:
Good, you'll also need to install it.

If you're on Ubuntu, you can use:

`time make -j$(nproc) package VERBOSE=1`

After the step above, and install the package's artifacts via:

`sudo dpkg -i *.deb`

Ran in the build directory.

You can install the package directly without generating Debian archives via:

`sudo make -j$(nproc) install VERBOSE=1`

Then proceed to build ICR. The build should complete without issues.

I should clarify the installation bit in my documentation.

Brainiarc7, 2018-10-20:
@dseeker can you confirm that installing ICR fixes the problem you're encountering?

dseeker, 2018-10-21:
Really thankful for helping out, but maybe I'm miscommunicating, because that's what doesn't work. when I do 
`time make -j$(nproc) package VERBOSE=1`
I get all the include errors

```
/root/intel-compute-runtime/workspace/neo/runtime/helpers/hw_info.h:9:10: fatal error: gtsysinfo.h: No such file or directory
 #include "gtsysinfo.h"
          ^~~~~~~~~~~~~
compilation terminated.
unit_tests/libult/CMakeFiles/igdrcl_libult_env.dir/build.make:62: recipe for target 'unit_tests/libult/CMakeFiles/igdrcl_libult_env.dir/__/base_object_allocator.cpp.o' failed
make[2]: *** [unit_tests/libult/CMakeFiles/igdrcl_libult_env.dir/__/base_object_allocator.cpp.o] Error 1
unit_tests/mocks/CMakeFiles/igdrcl_mocks.dir/build.make:134: recipe for target 'unit_tests/mocks/CMakeFiles/igdrcl_mocks.dir/mock_context.cpp.o' failed
make[2]: *** [unit_tests/mocks/CMakeFiles/igdrcl_mocks.dir/mock_context.cpp.o] Error 1
make[2]: Leaving directory '/root/intel-compute-runtime/workspace/build_icr'
CMakeFiles/Makefile2:6612: recipe for target 'unit_tests/mocks/CMakeFiles/igdrcl_mocks.dir/all' failed
unit_tests/libult/CMakeFiles/igdrcl_libult.dir/build.make:230: recipe for target 'unit_tests/libult/CMakeFiles/igdrcl_libult.dir/mock_gfx_family.cpp.o' failed
make[1]: *** [unit_tests/mocks/CMakeFiles/igdrcl_mocks.dir/all] Error 2
make[2]: *** [unit_tests/libult/CMakeFiles/igdrcl_libult.dir/mock_gfx_family.cpp.o] Error 1
In file included from /root/intel-compute-runtime/workspace/neo/runtime/command_stream/csr_definitions.h:10:0,
                 from /root/intel-compute-runtime/workspace/neo/runtime/command_stream/command_stream_receiver.h:9,
                 from /root/intel-compute-runtime/workspace/neo/runtime/command_stream/command_stream_receiver_hw.h:9,
                 from /root/intel-compute-runtime/workspace/neo/unit_tests/libult/ult_command_stream_receiver.h:9,
                 from /root/intel-compute-runtime/workspace/neo/unit_tests/libult/gen8.cpp:24:
/root/intel-compute-runtime/workspace/neo/runtime/helpers/hw_info.h:9:10: fatal error: gtsysinfo.h: No such file or directory
 #include "gtsysinfo.h"
          ^~~~~~~~~~~~~
compilation terminated.
unit_tests/libult/CMakeFiles/igdrcl_libult.dir/build.make:278: recipe for target 'unit_tests/libult/CMakeFiles/igdrcl_libult.dir/gen8.cpp.o' failed
make[2]: *** [unit_tests/libult/CMakeFiles/igdrcl_libult.dir/gen8.cpp.o] Error 1
make[2]: Leaving directory '/root/intel-compute-runtime/workspace/build_icr'
CMakeFiles/Makefile2:4135: recipe for target 'unit_tests/libult/CMakeFiles/igdrcl_libult.dir/all' failed
make[1]: *** [unit_tests/libult/CMakeFiles/igdrcl_libult.dir/all] Error 2
In file included from /root/intel-compute-runtime/workspace/neo/runtime/built_ins/sip.h:9:0,
                 from /root/intel-compute-runtime/workspace/neo/runtime/built_ins/built_ins.h:11,
                 from /root/intel-compute-runtime/workspace/neo/runtime/helpers/kernel_commands.h:9,
                 from /root/intel-compute-runtime/workspace/neo/runtime/helpers/timestamp_packet.h:10,
                 from /root/intel-compute-runtime/workspace/neo/runtime/command_queue/command_queue.h:13,
                 from /root/intel-compute-runtime/workspace/neo/unit_tests/command_queue/command_queue_fixture.h:11,
                 from /root/intel-compute-runtime/workspace/neo/unit_tests/command_queue/command_queue_fixture.cpp:8:
/root/intel-compute-runtime/workspace/neo/runtime/helpers/hw_info.h:9:10: fatal error: gtsysinfo.h: No such file or directory
 #include "gtsysinfo.h"

```

I tried continuing the process all to the end anyway but FFMPEG build also fails with

```
 ~/ffmpeg_sources/FFmpeg # PATH="$HOME/bin:$PATH" make -j$(nproc)
POD     doc/ffprobe.pod
HTML    doc/ffprobe.html
HTML    doc/ffplay.html
HTML    doc/ffmpeg.html
HTML    doc/ffplay-all.html
HTML    doc/ffmpeg-all.html
HTML    doc/ffprobe-all.html
POD     doc/ffmpeg.pod
POD     doc/ffplay.pod
POD     doc/ffmpeg-all.pod
POD     doc/ffplay-all.pod
POD     doc/ffprobe-all.pod
LD      ffmpeg_g
LD      ffplay_g
LD      ffprobe_g
MAN     doc/ffmpeg.1
MAN     doc/ffplay.1
MAN     doc/ffprobe.1
MAN     doc/ffmpeg-all.1
MAN     doc/ffplay-all.1
MAN     doc/ffprobe-all.1
libavformat/libavformat.a(rtsp.o): In function `ff_rtsp_connect':
[... 127 lines cut ...]
dsd.c:(.text+0x20): undefined reference to `ff_reverse'
libavcodec/libavcodec.a(aacps_fixed.o): In function `ff_ps_read_data_fixed':
aacps_fixed.c:(.text+0x3dcc): undefined reference to `ff_log2_tab'
libavcodec/libavcodec.a(aacps_float.o): In function `ff_ps_read_data':
aacps_float.c:(.text+0x3f4c): undefined reference to `ff_log2_tab'
libavformat/libavformat.a(matroskadec.o): In function `ebml_read_num.isra.3':
matroskadec.c:(.text+0x7bd): undefined reference to `ff_log2_tab'
libavformat/libavformat.a(matroskadec.o): In function `matroska_ebmlnum_uint':
matroskadec.c:(.text+0xfac): undefined reference to `ff_log2_tab'
libavformat/libavformat.a(matroskadec.o): In function `ebml_parse':
matroskadec.c:(.text+0x2697): undefined reference to `ff_log2_tab'
libavcodec/libavcodec.a(bitstream.o): In function `build_table':
bitstream.c:(.text+0x136): undefined reference to `ff_reverse'
bitstream.c:(.text+0x2df): undefined reference to `ff_reverse'
libavcodec/libavcodec.a(bitstream.o): In function `ff_init_vlc_sparse':
bitstream.c:(.text+0xacc): undefined reference to `ff_reverse'
bitstream.c:(.text+0x1114): undefined reference to `ff_reverse'
libavcodec/libavcodec.a(s302m.o): In function `s302m_decode_frame':
s302m.c:(.text+0x14f): undefined reference to `ff_reverse'
libavcodec/libavcodec.a(s302m.o):s302m.c:(.text+0x22c): more undefined references to `ff_reverse' follow
collect2: error: ld returned 1 exit status
Makefile:108: recipe for target 'ffplay_g' failed
make: *** [ffplay_g] Error 1

```

ganxiao2008, 2018-11-07:
Same issue here after exact step follow https://gist.github.com/Brainiarc7/4f831867f8e55d35cbcb527e15f9f116
From https://github.com/intel/compute-runtime/issues/61#issuecomment-418420052 we might use media-driver 18.3.pre2 pre-release instead