intel/media-driver #248: pkgconfig for igdgmm is lacking Libs options

chivakker, 2018-07-25:
Please include Libs information for proper linking from outside this library.

oliver-sang, 2018-07-25:
we will change to use dynamic gmmlib and regard gmmlib as an external dep. So we won't pursue to supply igdgmm pkgconfig information from media side.
Will close this issue when we finish the transition. Thanks

dvrogozh, 2018-08-08:
This is not media-driver issue! This is gmmlib issue and should be filed appropriately!

See: intel/gmmlib#26 (igdgmm.pc file should be corrected)

drprajap, 2018-08-09:
This issue has been addressed by gmmlib commit - [d3e1b05](https://github.com/intel/gmmlib/commit/d3e1b053f6639efc25959b0485a0f3def2833354).

Thanks.

yangnin, 2018-08-16:
This issue has been addressed by gmmlib commit - d3e1b05.

closed 2018-08-16 by yangnin