intel/media-driver #250: iHD_drv_video.so has no function __vaDriverInit_1_0

djmv, 2018-07-25:
I built and installed intel-vaapi-driver and Intel libva but I got this error with vainfo.

My computer:
model name: Intel® Core™ i7-4770 CPU @ 3.40GHz

vainfo
libva info: VA-API version 1.1.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva error: /usr/local/lib/dri/iHD_drv_video.so has no function __vaDriverInit_1_0
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit

[image]

dvrogozh, 2018-07-25:
intel-vaapi-driver is another driver actually: https://github.com/intel/intel-vaapi-driver

Please, provide details on what you have cloned, how configured, how installed and on which operating system you've done all the above. There is some inconsistency between versions, but I don't catch it from your initial message:(.

XinfengZhang, 2019-08-28:
@djmv is the issue still there, if no , I will close the issue in 1 week

XinfengZhang, 2019-08-29:
root cause:  please use https://github.com/intel/intel-vaapi-driver instread of media-driver, i7-4770 CPU is HSW, right? media-driver's support is from BDW

closed 2020-03-08 by XinfengZhang