intel/media-driver #251: [ffmpeg-qsv] [vp8-dec] it will occurs gpu hang and decoding md5 does not match the standard reference with commit 732e7e12090a2d43db12b9eb0eff0d05e8a15b5f

wangzj0601, 2018-07-26:
env:
platform: skl
media-driver: 732e7e12090a2d43db12b9eb0eff0d05e8a15b5f 
ffmpeg-qsv: commit 489486b4609ee131fbcd25b846ea4a8ef1c7addb, branch intel_ffmpeg-3.4.1
                    repo https://github.com/Intel-FFmpeg-Plugin/Intel_FFmpeg_plugins.git
mediasdk: commit fd6053573ebf7efc99d092e6dc02b523ef86b01e
                 repo https://github.com/Intel-Media-SDK/MediaSDK

step:
1. build env as above
2. ffmpeg -hwaccel qsv -hwaccel_device /dev/dri/renderD128 -v verbose -y -c:v vp8_qsv -i /media//vp8_vpg/vp8_rev0_I_frame_TM.webm -vf hwdownload,format=nv12 -pix_fmt yuv420p -f md5 qsv_out.md5
3. it will occurs gpu hang and the md5 is different with golden reference md5

wangzj0601, 2018-07-26:
it should be caused by commit 732e7e12090a2d43db12b9eb0eff0d05e8a15b5f

-    //macroblock_offset is in unit of bit.
-    picParams->uiFirstMbByteOffset = slcParam->slice_data_offset + slcParam->macroblock_offset >> 3;
+    //macroblock_offset is in unit of bit.it should be always the next byte, the byte is divided to two parts
+    //used bits and remaining bits, if used bits == 8, uiFirstMbByteOffset should add 1, so use 8 to do the ceil operator
+    picParams->uiFirstMbByteOffset = slcParam->slice_data_offset + ((slcParam->macroblock_offset + 8) >> 3);

XinfengZhang, 2018-07-29:
need #239 and related MSDK changes

eero-t, 2018-11-16:
Linked fixes seem to have been merged to master in September.  Can this be closed now, or didn't they fix this issue?

uartie, 2018-11-16:
WFM

eero-t, 2018-11-27:
VP8 webm files generated by Chromium, can still GPU hang when decoded. Pull request https://github.com/intel/media-driver/pull/430 should fix that.

closed 2019-08-28 by XinfengZhang