intel/media-driver #252: [SKL][AVC DEC] fate-h264-lossless test cases failed When Using ffmpeg-vaapi

mypopydev, 2018-07-26:
command:

./ffmpeg -nostdin -nostats -cpuflags all -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_flags allow_profile_mismatch -threads 1 -thread_type frame+slice -i ./fate-suite//h264/lossless.h264 -pix_fmt yuv420p -flags +bitexact -fflags +bitexact -f framecrc -

Get the output as below, can't match with the SW decoder

--- ./tests/ref/fate/h264-lossless	2018-06-07 14:32:26.688501113 +0800
+++ tests/data/fate/h264-lossless	2018-07-10 16:39:20.176783101 +0800
@@ -3,13 +3,13 @@
#codec_id 0: rawvideo
#dimensions 0: 640x480
#sar 0: 0/1
-0, 0, 0, 1, 460800, 0x7731dd2f
-0, 2, 2, 1, 460800, 0x944b8c64
-0, 3, 3, 1, 460800, 0xbe833041
-0, 4, 4, 1, 460800, 0xbe95d96a
-0, 5, 5, 1, 460800, 0xfe7ea5e6
-0, 6, 6, 1, 460800, 0x381743c7
-0, 7, 7, 1, 460800, 0x63fcc2e9
-0, 8, 8, 1, 460800, 0x79574960
-0, 9, 9, 1, 460800, 0xdab9e18a
-0, 10, 10, 1, 460800, 0xd88e8fe8
+0, 0, 0, 1, 460800, 0x4b65b7ac
+0, 2, 2, 1, 460800, 0xf257dc01
+0, 3, 3, 1, 460800, 0x48767109
+0, 4, 4, 1, 460800, 0xc5db95ec
+0, 5, 5, 1, 460800, 0x011141c5
+0, 6, 6, 1, 460800, 0x3b79ba35
+0, 7, 7, 1, 460800, 0xe87b24ec
+0, 8, 8, 1, 460800, 0xad8fbdb1
+0, 9, 9, 1, 460800, 0x80a258a6
+0, 10, 10, 1, 460800, 0x03528f17
Test h264-lossless failed. Look at tests/data/fate/h264-lossless.err for details.

XinfengZhang, 2020-03-08:
@wangyan-intel , please help to check whether the issue still there, from the comments, I suppose just because of the output surfaces size?

wangyan-intel, 2020-03-08:
@XinfengZhang I will check it.

wangyan42164, 2020-05-19:
@XinfengZhang @mypopydev 
This clip is h264 (High 4:4:4 Predictive). But current vaapi only support YUV420 format whatever H264 main/high/ ConstrainedBaseline profile. 
FFMPEG just select VAProfileH264High but it should be wrong.

wangyan-intel, 2020-05-19:
Based on the previous comments, this clip isn't supported by vaapi.

closed 2020-05-19 by wangyan-intel