intel/media-driver #258: [RFC] is it have support sliding Windows Bitrate control in media-driver

mypopydev, 2018-08-01:
Now MSDK have enabled support for sliding window constraints in CBR and VBR modes of the AVC
encoder (https://software.intel.com/sites/default/files/managed/9b/7b/mediasdkembedded_release_notes.pdf), is it have plan to support  sliding Windows Bitrate control in libva(VA-API)/ media-driver ?

XinfengZhang, 2018-08-01:
yes, using frame_tolerance_mode in VAEncMiscParameterRateControl can enable/disable it. but does not support sliding window size setting in VAAPI. for our driver , you can change it by registry file. by default it is 30 frames

XinfengZhang, 2018-08-02:
@mypopydev yes, add it into VAAPI interface maybe a good choice, but we suggest use default window size for most of popular cases.

closed 2019-08-28 by XinfengZhang