intel/media-driver #264: iHd with gstreamer-vaapi on multi GPU system (igpu intel and amd GPU) not working

arpu, 2018-08-03:
this works fine if i can set the device
 LIBVA_DRIVER_NAME=iHD vainfo --display drm --device /dev/dri/renderD129
libva info: VA-API version 1.2.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_2
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.2 (libva 2.2.1.pre1)
vainfo: Driver version: Intel iHD driver - 2.0.0
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD

but 
LIBVA_DRIVER_NAME=iHD gst-launch-1.0 v4l2src   device=/dev/video0  extra-controls=\"c,exposure_auto=1,exposure_absolute=63,gain=2\"  !  image/jpeg, framerate=60/1, width=1920, height=1440  ! vaapijpegdec  ! fpsdisplaysink name=fpssink text-overlay=true video-sink=xvimagesink signal-fps-measurements=true -v
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
failed

any idea how i can get gstreamer-vaapi work with iHD driver?

arpu, 2018-08-03:
i think i found the problem i have to set 

 export GST_VAAPI_ALL_DRIVERS=1

to use other drivers

arpu, 2018-08-03:
but than i get this error on vaapijpegenc

 LIBVA_DRIVER_NAME=iHD gst-launch-1.0 v4l2src   device=/dev/video0  extra-controls=\"c,exposure_auto=1,exposure_absolute=63,gain=2\"  !  image/jpeg, framerate=60/1, width=1920, height=1440  ! vaapijpegdec  ! fpsdisplaysink name=fpssink text-overlay=true video-sink=xvimagesink signal-fps-measurements=true -v
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
Setting pipeline to PAUSED ...
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstXvImageSink:xvimagesink0: sync = true
Got context from element 'vaapidecode_jpeg0': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayX11\)\ gldisplayx11-0";
Got context from element 'vaapidecode_jpeg0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayDRM\)\ vaapidisplaydrm1";
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = image/jpeg, framerate=(fraction)60/1, width=(int)1920, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)NULL, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, framerate=(fraction)60/1, width=(int)1920, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)NULL, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstVaapiDecode_jpeg:vaapidecode_jpeg0.GstPad:sink: caps = image/jpeg, framerate=(fraction)60/1, width=(int)1920, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)NULL, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = image/jpeg, framerate=(fraction)60/1, width=(int)1920, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)NULL, interlace-mode=(string)progressive
ERROR: from element /GstPipeline:pipeline0/GstVaapiDecode_jpeg:vaapidecode_jpeg0: Decoding error
Additional debug info:
../../../gst/vaapi/gstvaapidecode.c(764): gst_vaapidecode_handle_frame (): /GstPipeline:pipeline0/GstVaapiDecode_jpeg:vaapidecode_jpeg0:
Decode error -1
Execution ended after 0:00:01.032433718
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

arpu, 2018-08-03:
is there a environment variable for setting the device ?

XinfengZhang, 2018-08-10:
suppose, it is because /dev/dri/renderD129, if intel gpu is /dev/dri/renderD128, it should works, right?

arpu, 2018-08-17:
yes  i think it uses the first device right

arpu, 2018-08-17:
https://github.com/intel/intel-vaapi-driver/issues/186#issuecomment-413949304

looks like the same problem

Kabouik, 2018-08-17:
Indeed. All I would need would be to be able to set /dev/dri/render128 as my VA device with an environment variable, this way I could just `export thatvariable` and then run the application I want to use with i965. Have you found a way to select the device as `vainfo` allows with `--device` @arpu ?

jackersson, 2019-05-28:
Hi, 

Had anyone solved previous problem? (Having the same troubles)

cqian2, 2022-11-03:
Close the issue as old-timed.
Please the submitter help submit new issue if similar issue was got during recent dev progress.

closed 2022-11-03 by cqian2