intel/media-driver #267: [ICLLP][AVC Encode] AVC encode triggers GPU hang on ICLLP

zxye, 2018-08-08:
1) build and install https://github.com/intel/media-driver
2) build https://github.com/intel/libva-utils
3) $ libva-utils/encode/avcenc 640 480 ./test.yuv test.h264 qp=25 mode=1
libva info: VA-API version 1.2.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_2
libva info: va_openDriver() returns 0

In dmesg:
......
[1138846.694198] i915 0000:00:02.0: Resetting chip for hang on rcs0
[1138854.694139] i915 0000:00:02.0: Resetting chip for hang on vcs0

XinfengZhang, 2018-08-10:
thanks for reporting, will check it ASAP

walter-bai, 2018-08-10:
Is this issue a regression?

XinfengZhang, 2018-08-10:
@walter-bai, we just upstream ICLLP.  so , please help to check

dvrogozh, 2018-08-10:
This is not a regression. ICLLP requires media-driver to shutdown subslices which miss VME, i.e. it needs to use i915 uAPI considered here: https://patchwork.freedesktop.org/series/42285/. Mind that this uAPI is not yet available in a stock kernel. On media-driver side the use of this uAPI may need to be activated via:
```
# vi /etc/igfx_user_feature.txt
[KEY]
        0x00000001
        UFKEY_INTERNAL\
                [VALUE]
                        Slice Count Set Support
                        4
                        1
```
However, I am not 100% sure that this will fully activate the use of uAPI (@zxye: can you, please, comment: what is needed to activate the use of required i915 uAPI?). Also, mind that media-driver may need to be updated to use correct ioctl number from the mentioned patch...

zxye, 2018-08-12:
The patch to shut down non-VME subslices for ICLLP is not merged to master branch yet. You can find it here: https://github.com/zxye/media-driver/commit/74a75f0e9f804fa39171d414439f877e713b1afa

 https://patchwork.freedesktop.org/series/42285/ is not the final version of the i915 series. Once the i915 uAPI is finalized, we need to update the ioctl number and the definition of struct drm_i915_gem_context_param, rework mos_set_context_param_sseu(), and then merge 74a75f0e9f804fa39171d414439f877e713b1afa.

dvrogozh, 2018-08-12:
@zxye : thank you. Yes, i915 series is not finalized. However, we may use it in the current state and try to finalize media-driver change as much as we can. Can you, please, create a PR to media-driver from the commit you have noted above? Please, add "Fixes #267" in the commit message to the patch to link the PR to the current issue. I know that we can't merge in this PR before i915 series, but creating it will be in full accordance with kernel drm maintainers guidelines. That's really important, please, do.

I still don't quite understand one thing: does your patch require adding "Slice Count Set Support" config to /etc/igfx_user_feature.txt or on the ICLLP this uAPI will be used always after your change zxye@74a75f0?

dvrogozh, 2018-08-13:
@zxye : Thank you! No, on ICLLP config is not needed and feature simply should be used otherwise we will have a GPU hang.

zxye, 2018-08-21:
Submitted https://bugs.freedesktop.org/show_bug.cgi?id=107634 @fdo.

closed 2019-02-26 by Sherry-Lin

xuiqzy, 2019-10-06:
I'm not familiar with this bug, but since it is closed here, should the text on the [bottom of the README.md](https://github.com/intel/media-driver/blob/master/README.md), where it is referenced as an open bug, be changed?

commit 50277168 (2019-02-26): Set subslice mask according to the per-context sseu value. Fixes #267.