intel/media-driver #272: The vpp will ignore alpha value in output_background_color and always clear background to black.

xuguangxin, 2018-08-13:
According to libva definition, the output_background_color  is an ARGB value. If we set alpha to 0.  It will use the original background value. This feature is very useful if we want to draw mutli video on different areas. But the current driver will ignore the alpha and always clear the background to black. 

How to reproduce:
1. follow https://github.com/intel/libyami/wiki/Build to build libyami and the media-driver
2. configure libyami-utils using --enable-dma-buf
3. kill x server
4. run libyami-utils/exmpale/grid xxx.264 -r 4 -c 5

You will see only right conner has a video playback. other areas are black

Expected behaviors:
We can see 4x5(20) video playback on screen

FurongZhang, 2018-08-14:
Thanks for the details of issue. Will try to reproduce the issue first.

xuguangxin, 2018-08-24:
hi @FurongZhang , any update on this? thanks

Jister, 2018-08-30:
Hi, Guangxin, I have create a pull request to fix the issue, please refer to #296

PR #324 (2018-10-18): [VP]When alpha channel of background color is 0, will not do colorfill.
Fix #272 , previously, color fill parameter will always be allocated, here add the condition when alpha channel of background color is not 0
closed 2018-10-18 by Sherry-Lin