intel/media-driver #276: Implementation of vaCreateSurface contradicts va.h

dmitryermilov, 2018-08-14:
According to va.h semantic of vaCreateSurface is the following:

```
/**
 * \brief Creates an array of surfaces
 *
 * Creates an array of surfaces. The optional list of attributes shall
 * be constructed based on what the underlying hardware could expose
 * through vaQuerySurfaceAttributes().
 *
 * @param[in] dpy               the VA display
 * @param[in] format            the desired surface format. See \c VA_RT_FORMAT_*
 * @param[in] width             the surface width
 * @param[in] height            the surface height
 * @param[out] surfaces         the array of newly created surfaces
 * @param[in] num_surfaces      the number of surfaces to create
 * @param[in] attrib_list       the list of (optional) attributes, or \c NULL
 * @param[in] num_attribs       the number of attributes supplied in
 *     \c attrib_list, or zero
 */
VAStatus
vaCreateSurfaces(
    VADisplay           dpy,
    unsigned int        format,
    unsigned int        width,
    unsigned int        height,
    VASurfaceID        *surfaces,
    unsigned int        num_surfaces,
    VASurfaceAttrib    *attrib_list,
    unsigned int        num_attribs
);
```

Pay attention to the argument "format" :
`[in] format            the desired surface format. See \c VA_RT_FORMAT_*`

It's has VA_RT_FORMAT_*** type.

Then, please look at current implementation of vaCreateSurface:
https://github.com/intel/media-driver/blob/36b3a0aa3b47a4cda36a31d98e16da9ff7ea27fb/media_driver/linux/common/ddi/media_libva.cpp#L1946-L2045

It checks in the same switch block VA_RT_FORMAT_*** and VA_FOURCC_*** which doesn't look right. 
VA_FOURCC_*** should be read via attrib_list when type is VASurfaceAttribPixelFormat:
```
/** \brief Surface attribute types. */
typedef enum {
    VASurfaceAttribNone = 0,
    /**
     * \brief Pixel format (fourcc).
     *
     * The value is meaningful as input to vaQuerySurfaceAttributes().
     * If zero, the driver returns the optimal pixel format for the
     * specified config. Otherwise, if non-zero, the value represents
     * a pixel format (FOURCC) that is kept as is on output, if the
     * driver supports it. Otherwise, the driver sets the value to
     * zero and drops the \c VA_SURFACE_ATTRIB_SETTABLE flag.
     */
VASurfaceAttribPixelFormat,
```

XinfengZhang, 2018-08-16:
sure, we know this , it is just for some application

dmitryermilov, 2018-08-16:
I'm sure you know:) The question is do libva users know how to correctly write applications taking into this contradiction?

XinfengZhang, 2018-08-28:
we will add comments inside driver to describe it, customer should use attrib list to add specific fourcc, or it will be default by the RT_FORMAT

closed 2019-08-29 by XinfengZhang