intel/media-driver #279: [ffmpeg-vaapi][MPEG2_encoding encode] the MPEG2_encoding encode failed on CFL/KBL/SKL platforms by using ffmpeg-vaapi

zhuqingliang, 2018-08-16:
Test Env:
======================

Platform: CFL/KBL/SKL
Arch: x86_64
Kernel   version: CFL 4.15.0,    KBL  4.14.0-rc1 , SKL  4.15.0
Libva:   6ae717304c48e7710a76a87cef2a1096ff450af2
media_driver:   36b3a0aa3b47a4cda36a31d98e16da9ff7ea27fb
libva-utils:  9a10ad663349732decd668a426cfc349b0010d1d
ffmpeg-vaapi:   5423fe29da6bbf28f5253800cace133ac8343a7e(master branch)

Reproduce Steps:
=========================

1. build enc as above lists
2.  ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -y -s:v 2048x858 -i /root/media_streams//encoderbitstreams//Gravity-2K_Trailer_300frames.yuv -vf format=nv12,hwupload -c:v mpeg2_vaapi -profile:v 4 -level 10 -g 59 -bf 0 -global_quality 5 2k_858p_i420_profile4_level10_bf0_gop59_gquality5_fps30.m2v
3. error message
[mpeg2_vaapi @ 0x1ac66c0] Warning: some packed headers are not supported (want 0x3, got 0).
[mpeg2_vaapi @ 0x1ac66c0] Hardware does not support encoding at size 2048x864 (constraints: width 32-1920 height 32-1920).
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
Conversion failed!

Test case:
==================================
2k_858p_i420_profile4_level10_bf0_gop59_gquality5_fps30.m2v
2k_858p_i420_profile4_level8_bf1_gop30_gquality10_fps30.m2v
2k_858p_i420_profile4_level4_bf0_gop59_gquality20_fps30.m2v
2k_858p_i420_profile4_level4_bf1_gop30_gquality5_fps30.m2v
2k_858p_i420_profile4_level8_bf0_gop59_gquality10_fps30.m2v
2k_858p_i420_profile5_level10_bf0_gop59_gquality5_fps30.m2v
2k_858p_i420_profile5_level8_bf1_gop30_gquality10_fps30.m2v
2k_858p_i420_profile5_level4_bf0_gop59_gquality20_fps30.m2v
2k_858p_i420_profile5_level4_bf1_gop30_gquality5_fps30.m2v
2k_858p_i420_profile5_level8_bf0_gop59_gquality10_fps30.m2v
4K_2160p_i420_profile4_level10_bf0_gop59_gquality5_fps30.m2v
4K_2160p_i420_profile4_level8_bf1_gop30_gquality10_fps30.m2v
4K_2160p_i420_profile4_level4_bf0_gop59_gquality20_fps30.m2v
4K_2160p_i420_profile4_level4_bf1_gop30_gquality5_fps30.m2v
4K_2160p_i420_profile4_level8_bf0_gop59_gquality10_fps30.m2v
4K_2160p_i420_profile5_level10_bf0_gop59_gquality5_fps30.m2v
4K_2160p_i420_profile5_level8_bf1_gop30_gquality10_fps30.m2v
4K_2160p_i420_profile5_level4_bf0_gop59_gquality20_fps30.m2v
4K_2160p_i420_profile5_level4_bf1_gop30_gquality5_fps30.m2v
4K_2160p_i420_profile5_level8_bf0_gop59_gquality10_fps30.m2v

XinfengZhang, 2018-08-28:
@jcescamilla , you will add packed header support, right?

XinfengZhang, 2018-08-28:
driver does not support packed header insertion for MPEG2, now it only support raw_data. not support PPS, SliceHeader etc.

Xiaogangli-intel, 2019-09-03:
@jcescamilla Any other questions on this issue? If no, I will close this issue.

closed 2020-05-14 by XinfengZhang