intel/media-driver #281: [ffmpeg-vaapi][mjpeg enc]the quality of encoded file with option "global_quality" level 90 is lower than the level 50

wangzj0601, 2018-08-17:
Test Env:
Platform: CFL/KBL/SKL
Arch: x86_64
Kernel version: CFL 4.15.0,    KBL  4.14.0-rc1 , SKL  4.15.0
Libva: 6ae717304c48e7710a76a87cef2a1096ff450af2
media_driver: 36b3a0a
libva-utils:  9a10ad663349732decd668a426cfc349b0010d1d
ffmpeg-vaapi: 5423fe29da6bbf28f5253800cace133ac8343a7e(master branch)

Test step:
1. ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -y -s:v 1920x1080 -i /root/media_streams//encoderbitstreams//1080p_blue_sky.yuv -vf format=nv12,hwupload -c:v mjpeg_vaapi -global_quality 90 1080p_1920x1080_i420_gquality90_fps30.mp4
2. ffmpeg -i ./1080p_1920x1080_i420_gquality90_fps30.mp4 -pix_fmt yuv420p dump.yuv
3. metrics_calc_lite -i1 /root/media_streams//encoderbitstreams//1080p_blue_sky.yuv -i2 dump.yuv -w 1920 -h 1080 -nopfm psnr y
4. YSNR=' 12.41936', and play the encoded file, find that the quality is very low
5. ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -y -s:v 1920x1080 -i /root/media_streams//encoderbitstreams//1080p_blue_sky.yuv -vf 'format=nv12,hwupload' -c:v mjpeg_vaapi -global_quality 50 1080p_1920x1080_i420_gquality50_fps30.mp4
7. YSNR=' 27.85109'

XinfengZhang, 2018-08-28:
we have a fix about the quality setting. https://github.com/intel/media-driver/pull/289/commits/e1345456a4a378ba19da80e47f77f2be5ad4e4d4

wangzj0601, 2018-09-13:
Does this commit has been merged into master branch? This issue still can be duplicated with the latest driver
commit d32d5ddc9daed9dd40a7fc22288a9ab3f1788d10
Author: xiong_(sean)__zhou <sean.zhou@intel.com>
Date:   Thu Sep 13 09:14:06 2018 +0800

global_quality | bitrate | psnr
10 | 11699.9 | 33.17715
20 | 15855.4 | 36.84325
30 | 19859.5 | 36.71817
40 | 21729.7 | 39.76352
50 | 25826.1 | 41.36427
60 | 27469.1 | 33.66301
70 | 38617.9 | 31.47269
80 | 49228.7 | 40.64858
90 | 63495.7 | 41.69412
100 | 133950.9 | 63.01732

global_quality 60/70/80 should be higher than setting 50.

wangzj0601, 2018-12-05:
this issue also can be duplicate on gstreamer-vaapi, cmdline as below
gst-launch-1.0 filesrc location=/media/encoderbitstreams//720x480p.yuv '!' videoparse format=i420 width=720 height=480 '!' videoconvert '!' video/x-raw,format=NV12 '!' vaapijpegenc quality=10 '!' filesink location=720x480p.yuv.jpeg
result:
720x480p.yuv_720_480_10  29.73393 PASS
720x480p.yuv_720_480_20  32.22547 PASS
720x480p.yuv_720_480_40  34.87364 PASS
720x480p.yuv_720_480_60  34.30753 FAIL
720x480p.yuv_720_480_80  36.15495 FAIL

wangzj0601, 2019-02-21:
now this issue can be duplicated on jpeg encoding while pass on mjpeg encoding with ffmpeg-vaapi/ffmpeg-qsv/gst-vaapi/gst-msdk.

FocusLuo, 2019-09-17:
@XinfengZhang any update on this issue?

xhaihao, 2019-11-25:
This issue can be duplicated by using sample_encode in https://github.com/Intel-Media-SDK/MediaSDK, so this is not a ffmpeg vaapi specific issue.

xhaihao, 2020-03-06:
@XinfengZhang any update for this issue?

XinfengZhang, 2020-03-08:
@xhaihao , the issue still there after fix the repeated scaled matrix issue?

Alexandr-Konovalov, 2020-05-26:
@wangzj0601, @xhaihao Could you please share current command lines to reproduce the issue? Quality drop from the original message looks resolved.

XinfengZhang, 2020-05-26:
@Alexandr-Konovalov , sample_encode also could reproduce the issue

Alexandr-Konovalov, 2020-05-26:
So, the current issue is a lack of monotonity in quality range 50 - 80? OK, got it.

```
 q | dB
90 | 28.62462
80 | 28.52039
70 | 28.45082
60 | 28.47275
50 | 28.50397
40 | 28.27968
```

Alexandr-Konovalov, 2020-05-27:
It seems that lack of monotony in wrt quality increase is a result of how jpeg designed, not a bug of some implementation. I.e., for same 1st frame of 1080p_blue_sky.yuv in  libjpeg-turbo I got
```
90 | 28.03798
80 | 28.02770
70 | 28.37393
60 | 28.22193
50 | 28.22758
```
It’s unclear what can be done here.

Alexandr-Konovalov, 2020-06-03:
@wangzj0601, Are you OK with closing it, or some more analysis required?

closed 2021-01-31 by XinfengZhang