intel/media-driver #287: Additional Image formats

krogueintel, 2018-08-24:
The older intel-vaapi-driver supported the the following formats: : IA44, AI44, IA88, AI88, ARGB, ABGR,, 422H, and UYVY which the intel media-driver does not. For better transitioning, would it be possible to support these formats as well?

XinfengZhang, 2018-08-28:
yes,we add more Image support , base on the format driver support (VPP&CODEC)

FurongZhang, 2019-02-16:
@Jister , Have we merged the fix? Is it possible for you to update the status? If merged, how about closing this issue? Thanks a lot!

Jister, 2019-02-18:
IA44, AI44, IA88 will not be supported, unless there is specific/strong requirement. Other formats have be added.

FurongZhang, 2019-02-18:
@krogueintel  Have you had a chance to check if this issue is fixed or not? Thanks a lot in advance! Currently, we don't have a plan to support IA44, AI44, IA88 until there is a strong request since they are legacy formats. 
BTW, do you know if there is some main usage for IA44, AI44, IA88 formats?

krogueintel, 2019-03-04:
I changed groups; any work on the media driver is completely out of scope (and indeed almost unrelated) for my group's responsibilities.

FurongZhang, 2019-03-21:
Thanks @krogueintel for the information.

closed 2019-08-28 by XinfengZhang