intel/media-driver #288: vaapisink doesn't keep aspect ratio

riccardocagnasso, 2018-08-26:
Regardless of the options about aspect ration on vaapisink and vaapipostproc, a video played with the iHD codec will stretch to occupy all the output window.

[pipeline.zip](https://github.com/intel/media-driver/files/2321336/pipeline.zip)
here is an example pipeline

XinfengZhang, 2018-08-28:
we will follow up the issue, but we suggest to use glimagesink

Jister, 2018-09-12:
@riccardocagnasso Could you please provide an example case for us to reproduce the issue?

riccardocagnasso, 2018-09-21:
@Jister yes. pick a video that has a different aspect ration that of your monitor, then

gst-launch-1.0 filesrc location=video.mp4 ! qtdemux !  vaapidecodebin ! vaapisink fullscreen=true

Jister, 2018-09-26:
@riccardocagnasso I create a pull request #329, you can have a try whether it can fix your issue.

riccardocagnasso, 2018-10-06:
@Jister i cloned your whole forked repository but i got a failure to compile

```
/home/phas/Downloads/foo/src/intel-media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp: In member function ‘virtual VAStatus DdiMediaDecode::CreateBuffer(VABufferType, uint32_t, uint32_t, void*, VABufferID*)’:
/home/phas/Downloads/foo/src/intel-media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:745:14: error: ‘VASubsetsParameterBufferType’ was not declared in this scope
         case VASubsetsParameterBufferType:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/phas/Downloads/foo/src/intel-media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:745:14: note: suggested alternative: ‘VASliceParameterBufferType’
         case VASubsetsParameterBufferType:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
              VASliceParameterBufferType
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:5276: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_base.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:170: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

Not sure if this is related to this patch actually

Jister, 2018-10-07:
@riccardocagnasso It seems your Libva is not the correct version, please update Libva to lastest.

Jister, 2018-10-08:
@riccardocagnasso I have tried on my local side, there is no such build issue. Could you provide your operating system version?

riccardocagnasso, 2018-10-08:
@Jister ok i connected my brain and managed to compile the thing. But the bug seems to be still there.

[image]

Jister, 2018-10-09:
@riccardocagnasso Have you used the branch "ws1" from my repository?

riccardocagnasso, 2018-10-09:
@Jister ok, now I got it, sorry about the mess but I'm working in a messy environment at the moment.
Now the sink keeps the aspect ratio, but the positioning is off, see the attached screenshot.

[image]

Jister, 2018-10-10:
@riccardocagnasso I push another fix into branch ws1, could you please update the code and test it again, the output should be fine now.

Jister, 2018-10-10:
@riccardocagnasso OK, thanks. Will merge the fix into master branch later.

PR #329 (2018-10-18): [VP]Keep aspect ratio in PutSurface.
Fix #288
closed 2018-10-18 by Sherry-Lin